Zhefan-Xu/NavRL #1: Assets

gaohaowe, 2025-02-26:
Dr. Xu, hello. I have read your excellent open-source project and tried to reproduce it. When running the training script, I encountered an issue where assets could not be read from the cloud. I tried to download the assets folder to my local machine using a method similar to IsaacSim 4.5, but I couldn’t find the interface to import the assets folder.

StriverAlex, 2025-03-01:
Dear Dr.Xu, thanks for you excellent open-source project and I am learning it. I noticed the NavRL framework is built upon OmniDrones, can I use it based on higher version isaac sim through own edit？Thanks for your response.

Zhefan-Xu, 2025-03-01:
@StriverAlex Since the project is developed based on a earlier version of OmniDrones and we also made some modifications based on that, using the latest OmniDrones might not be directly feasible. But it is definitely doable if appropriate changes are made. Currently, I don't have a specific plan on migrating to the latest Isaac but we welcome people to open a pull request.