Zhefan-Xu/NavRL #4: ImportError: cannot import name 'RenderCallback' from 'omni_drones.utils.torchrl'

zwhhhhh9, 2025-03-04:
/home/action/Documents/VolleyBots/third_party/rl/torchrl/data/replay_buffers/samplers.py:37: UserWarning: Failed to import torchrl C++ binaries. Some modules (eg, prioritized replay buffers) may not work with your installation. If you installed TorchRL from PyPI, please report the bug on TorchRL github. If you installed TorchRL locally and/or in development mode, check that you have all the required compiling packages.
  warnings.warn(EXTENSION_WARNING)
Traceback (most recent call last):
  File "/home/action/Documents/NavRL/isaac-training/training/scripts/train.py", line 9, in <module>
    from ppo import PPO
  File "/home/action/Documents/NavRL/isaac-training/training/scripts/ppo.py", line 9, in <module>
    from utils import ValueNorm, make_mlp, IndependentNormal, Actor, GAE, make_batch, IndependentBeta, BetaActor, vec_to_world
  File "/home/action/Documents/NavRL/isaac-training/training/scripts/utils.py", line 7, in <module>
    from omni_drones.utils.torchrl import RenderCallback
ImportError: cannot import name 'RenderCallback' from 'omni_drones.utils.torchrl' (/home/action/Documents/VolleyBots/omni_drones/utils/torchrl/__init__.py)

Zhefan-Xu, 2025-03-04:
Hi @zwhhhhh9, could you please elaborate your issue?  For example, the command you ran for installation, errors from installation, and your installed NavRL venv

zwhhhhh9, 2025-03-04:
Ubuntu20.04 
cuda 11.7
python 3.10

zwhhhhh9, 2025-03-04:
Traceback (most recent call last):
  File "/home/action/Documents/NavRL/isaac-training/training/scripts/train.py", line 9, in <module>
    from ppo import PPO
  File "/home/action/Documents/NavRL/isaac-training/training/scripts/ppo.py", line 4, in <module>
    from tensordict.tensordict import TensorDict
  File "/home/action/Documents/NavRL/isaac-training/third_party/tensordict/tensordict/__init__.py", line 6, in <module>
    from tensordict._lazy import LazyStackedTensorDict
  File "/home/action/Documents/NavRL/isaac-training/third_party/tensordict/tensordict/_lazy.py", line 25, in <module>
    from tensordict._td import _SubTensorDict, _TensorDictKeysView, TensorDict
  File "/home/action/Documents/NavRL/isaac-training/third_party/tensordict/tensordict/_td.py", line 38, in <module>
    from tensordict.memmap import MemoryMappedTensor
  File "/home/action/Documents/NavRL/isaac-training/third_party/tensordict/tensordict/memmap.py", line 25, in <module>
    from torch.multiprocessing.reductions import ForkingPickler
ImportError: cannot import name 'ForkingPickler' from 'torch.multiprocessing.reductions' (/home/action/anaconda3/envs/NavRL/lib/python3.10/site-packages/torch/multiprocessing/reductions.py)

Zhefan-Xu, 2025-03-04:
Please take a look at this: https://github.com/pytorch/pytorch/issues/2797.

You might have an incompatiable pytorch version.

z1Xuan, 2025-03-20:
same problem：
$ python training/scripts/train.py
Traceback (most recent call last):
  File "/home/wzx/NavRL/isaac-training/training/scripts/train.py", line 9, in <module>
    from ppo import PPO
  File "/home/wzx/NavRL/isaac-training/training/scripts/ppo.py", line 9, in <module>
    from utils import ValueNorm, make_mlp, IndependentNormal, Actor, GAE, make_batch, IndependentBeta, BetaActor, vec_to_world
  File "/home/wzx/NavRL/isaac-training/training/scripts/utils.py", line 7, in <module>
    from omni_drones.utils.torchrl import RenderCallback
ModuleNotFoundError: No module named 'omni_drones'

SamforRL, 2025-06-02:
> same problem： $ python training/scripts/train.py Traceback (most recent call last): File "/home/wzx/NavRL/isaac-training/training/scripts/train.py", line 9, in from ppo import PPO File "/home/wzx/NavRL/isaac-training/training/scripts/ppo.py", line 9, in from utils import ValueNorm, make_mlp, IndependentNormal, Actor, GAE, make_batch, IndependentBeta, BetaActor, vec_to_world File "/home/wzx/NavRL/isaac-training/training/scripts/utils.py", line 7, in from omni_drones.utils.torchrl import RenderCallback ModuleNotFoundError: No module named 'omni_drones'

Hi, I encountered the same problem.May I ask if you have solved this problem？ @z1Xuan