Zhefan-Xu/NavRL #7: Running training/scripts/train.py reports an error

Rey-nard, 2025-03-08:
First of all, thanks for the great work!
I have followed the readme and finish the setup part, and successfully received the message "Setup completed successfully!"
and then I try to run the train.py scripts with "headless" set to be "True" to test whether it works well on my remote server.
But I have met the error below, do you have any idea how should fix the problem?

Traceback (most recent call last):
  File "/home/wrp/NavRL/isaac-training/training/scripts/train.py", line 49, in main
    env = NavigationEnv(cfg)
  File "/home/wrp/NavRL/isaac-training/third_party/rl/torchrl/envs/common.py", line 159, in __call__
    instance: EnvBase = super().__call__(*args, **kwargs)
  File "/home/wrp/NavRL/isaac-training/training/scripts/env.py", line 39, in __init__
    super().__init__(cfg, cfg.headless)
  File "/home/wrp/NavRL/isaac-training/third_party/OmniDrones/omni_drones/envs/isaac_env.py", line 134, in __init__
    global_prim_paths = self._design_scene() # in the child class, init robot (1), light, terrain (with number of env)
  File "/home/wrp/NavRL/isaac-training/training/scripts/env.py", line 100, in _design_scene
    cfg_ground.func("/World/defaultGroundPlane", cfg_ground, translation=(0, 0, 0.01))
  File "/home/wrp/NavRL/isaac-training/third_party/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/sim/spawners/from_files/from_files.py", line 145, in spawn_ground_plane
    collision_prim_path = prim_utils.get_prim_path(
  File "/mnt/data6/wrp/isaac-sim/exts/omni.isaac.core/omni/isaac/core/utils/prims.py", line 349, in get_prim_path
    return prim.GetPath()
AttributeError: 'NoneType' object has no attribute 'GetPath'

Zhefan-Xu, 2025-03-08:
Could you clarify what Isaac version are you using?

Rey-nard, 2025-03-09:
I'm using isaac-sim:2023.1.0-hotfix.1.

Zhefan-Xu, 2025-03-09:
This seems weird...The error suggests that the ground plane is not loaded properly which might be due to incompatible isaac version. I would recommend to try running on the local machine first. Please let me know if that works or not.

Rey-nard, 2025-03-09:
Thanks for your quick response, that's indeed a good idea to test it on the local machine first.

Rey-nard, 2025-03-10:
Besides, the ground plane is supposed to be saved locally or needs to be downloaded from the internet? If it's the latter case, could I download the resources in advance?

Rey-nard, 2025-03-10:
I downloaded the ground plane usd manually and modified the usd_path of the cfg_ground, isaac-sim still reminded me that certain assets weren't accessible, but the scene was loaded successfully. It seems that I have to download all the neccessary assets manually to avoid the network issue.  Since the complete asset pack available online is quite large and contains many unnecessary assets, making it relatively difficult to obtain. Could you please provide the assets actually used for training. Thanks in advance for your time!

[image]

Zhefan-Xu, 2025-03-10:
From the looking, it seems everything except the ground plane is loaded okay. Are you sure that the ground plane is properly loaded?

Rey-nard, 2025-03-11:
I'll check whether the ground plane is loaded normally. Besides, is it possible to provide the trained weight of the model?

Zhefan-Xu, 2025-03-11:
The trained model is included in both ros1/ros2 deployment and you can try them directly following the steps.

AJingshou, 2025-03-12:
Hey, could you please teach me how to download the usd?

WinstonJQ, 2025-04-04:
I am using the Isaac-Sim version 2023.1.0-hotfix.1. I encountered the same error when running train.py. Could you please share how you managed to resolve it? I would really appreciate your help. Thank you!
[image]

WinstonJQ, 2025-04-04:
> I am using the Isaac-Sim version 2023.1.0-hotfix.1. I encountered the same error when running train.py. Could you please share how you managed to resolve it? I would really appreciate your help. Thank you! [image]

When I enabled the network proxy, opened a new terminal, and ran train.py again, the issue was resolved. So, it seems that the error was caused by a network problem.

xch1176411706, 2025-05-26:
> I downloaded the ground plane usd manually and modified the usd_path of the cfg_ground, isaac-sim still reminded me that certain assets weren't accessible, but the scene was loaded successfully. It [...]

Hello, I'd like to know how you obtained the usd resources for this floor