Zhefan-Xu/NavRL #8: Minimum configuration requirements for local servers required for training agents

JinPaiGangC, 2025-03-11:
Hello Dr. Xu, first of all, thank you and your team for making this significant achievement. I really hope to replicate your project, but due to device limitations, may I ask if the local server graphics card configuration can be reduced? If the graphics card configuration can be reduced, what type of graphics card is required at least? And how should I modify the graphics card related configuration parameters in Isaac Sim training files? Looking forward to your response.

Zhefan-Xu, 2025-03-11:
Thanks for the comments. 
1. Yes, the minimum requirement install can be added. I think you only need to install the things in the setup.bash except ```Orbit``` and ```OmniDrones```. Also, you might need to install PyTorch first in the venv. If you can help test it out, I would be able to upload a minimum requirement version soon.
2. If you only want to run the ros1/ros2 deployment, you don't have to have a strong graphic card. You can just follow the steps to install the virtual env and run the deployment. The isaac installation does not need a strong GPU.

JinPaiGangC, 2025-03-11:
Thank you very much for your reply. I will try and get back to you.