Zhefan-Xu/NavRL #10: Issues with Drone Rendering During Training

zwhhhhh9, 2025-03-12:
When I run python training/script/train.py, I do not see two drones flying in the environment.

When I run the command:

python training/script/train.py headless=True env.num_envs=1024 env.num_obstacles=350 env_dyn.num_obstacles=80 wandb.mode=online

I do not see 1024 drones flying in the environment.

I can only see the moving obstacles.

[image]

Zhefan-Xu, 2025-03-12:
Thanks for posting the issue. You do see 1024 drones on the bottom (which are not moving because they are not well trained). After training for certain time, you will see them move around.

zwhhhhh9, 2025-03-12:
> Thanks for posting the issue. You do see 1024 drones on the bottom (which are not moving because they are not well trained). After training for certain time, you will see them move around.

How much GPU memory is required for your training? I am using an RTX 4080 with 16GB VRAM and I’m not sure if it’s enough to complete the training for your demo.

Zhefan-Xu, 2025-03-12:
> > Thanks for posting the issue. You do see 1024 drones on the bottom (which are not moving because they are not well trained). After training for certain time, you will see them move around.
> 
> Your YAML file defaults to two drones, but during training, I did not see these two drones in the simulation environment.

Could you clarify this question? The default is set to 2 for demonstration purpose and you should be able to see it on the bottom of the screen but can be small. I am a bit confused because I saw your screenshot having multiple drones on the bottom.

1024 drones should use RTX 4090 which has 24 GB. You can try using less number of drones.

zwhhhhh9, 2025-03-12:
> Could you clarify this question? The default is set to 2 for demonstration purpose and you should be able to see it on the bottom of the screen but can be small. I am a bit confused because I saw [...]

This screenshot was taken after my training session ended. As you can see, the 1024 drones did not move at all.

Zhefan-Xu, 2025-03-12:
How long did you train?

zwhhhhh9, 2025-03-12:
> 你训练了多久？

one night

Zhefan-Xu, 2025-03-12:
Do you mind sharing your wandb results?

zwhhhhh9, 2025-03-12:
> 您介意分享您的 wandb 结果吗？
Sorry, I deleted it and am now retraining.
[image]

Zhefan-Xu, 2025-03-12:
[image]

you should be able to see the videos every 1000 steps.

zwhhhhh9, 2025-03-12:
> [image]
> 
> you should be able to see the videos every 1000 steps.

There is only one video

[image]

zwhhhhh9, 2025-03-12:
> 这意味着由于某种原因您的训练停止了。请先尝试使用较少的无人机。它可能内存不足。
How should I modify the configuration to increase the training step and render the result every 1000 steps?
[image]

zwhhhhh9, 2025-03-14:
> 这意味着由于某种原因您的训练停止了。请先尝试使用较少的无人机。它可能内存不足。

How long does the training take with the default parameter settings?

Zhefan-Xu, 2025-03-14:
Approximate 10-15 hours on 4090