Zhefan-Xu/NavRL #13: Conda activate NavRL

Mming02, 2025-03-19:
Hello Dr. Xu, I encountered this error when executing conda activate NavRL: 
bash: /root/NavRL/isaac-training/third_party/orbit/_isaac_sim/setup_conda_env.sh: No such file or directory.    How can I resolve this?

VisionTrekker, 2025-03-19:
The possible cause is that the terminal used is `zsh` instead of bash. If you are using `zsh`, it's necessary to modify the terminal path in `_isaac_sim` for both `setup_conda_env.sh` and `setup_python_env.sh`.

Here are the modifications:

For `setup_conda_env.sh`, update:
`SCRIPT_DIR=$(cd "$(dirname "$0")" && pwd)`

For `setup_python_env.sh`, update:
`SCRIPT_DIR="$(dirname "$0")"`

> https://github.com/isaac-sim/IsaacLab/issues/103#issuecomment-1662112177

Mming02, 2025-03-19:
thanks , but i use bash.  I'm wondering if there is a problem with my isaac_sim-2023.1.0-hotfix.1.

VisionTrekker, 2025-03-19:
I use the my isaac_sim-2023.1.0-hotfix.1. in docker, and modify the terminal path. It works

Zhefan-Xu, 2025-03-19:
Could you check whether the path is correct "/root/NavRL/isaac-training/third_party/orbit/_isaac_sim/setup_conda_env.sh"? Also provide your testing environment

yuxuanlib, 2025-03-21:
> Hello Dr. Xu, I encountered this error when executing conda activate NavRL: bash: /root/NavRL/isaac-training/third_party/orbit/_isaac_sim/setup_conda_env.sh: No such file or directory. How can I resolve this?
Hi, to set the ISAACSIM_PATH using an absolute path, you can use the command echo 'export ISAACSIM_PATH="/your_path/
isaac_sim-2023.1.0-hotfix.1"' >> ~/.bashrc and then source ~/.bashrc to apply the change.

yuxuanlib, 2025-03-21:
Hi, to set the ISAACSIM_PATH using an absolute path, you can use the command echo 'export ISAACSIM_PATH="/your_path/
isaac_sim-2023.1.0-hotfix.1"' >> ~/.bashrc and then source ~/.bashrc to apply the change.