Zhefan-Xu/NavRL #14: env is incompatible

z1Xuan, 2025-03-20:
hello Dr.Xu
When I run bash setup.sh, torchrl 0.72 requires tensordict>=0.7.0 and torch>=2.6.0, but after I updated torch, the system reported an error that omni needs torch=2.0.1.
Running setup.py develop for omni-isaac-orbit
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
torchrl 0.7.2 requires tensordict>=0.7.0, but you have tensordict 0.2.1 which is incompatible.
torchrl 0.7.2 requires torch>=2.6.0, but you have torch 2.0.1 which is incompatible.

Zhefan-Xu, 2025-03-21:
Could you try torch==2.0.1 with tensordict and torchrl installed locally?

z1Xuan, 2025-03-21:
> Could you try torch==2.0.1 with tensordict and torchrl installed locally?

I removed the old conda environment and ran setup.sh, and these issues were resolved. But I found that when I run train.py, isaac-sim is always easy to stall, and then the process is killed, is this normal？ubuntu22.04LTS, RTX4080Laptop, 12GB

z1Xuan, 2025-03-21:
> > > 您能尝试在本地安装 tensordict 和 torchrl 的情况下使用 torch==2.0.1 吗？
> > 
> > 
> > 我删除了旧的 conda 环境，运行了 setup.sh，这些问题都解决了。但是我发现在运行 train.py 的时候，isaac-sim 总是容易卡住，然后进程就被杀死了，这正常吗？ubuntu22.04LTS，RTX4080Laptop，12GB
> 
> I’m curious whether it’s possible to train using Ubuntu 22.04.Theoretically, it is normal for Isaac Sim to be a bit slow or even freeze for a while during its first launch.

Which system version is the most suitable? I am currently trying to retrain on Ubuntu 20.04LTS and deploy NavRL with ROS1 and Gazebo.