Zhefan-Xu/NavRL #15: 运行bash setup.sh

wei-02, 2025-03-22:
运行bash setup.sh的时候，报错pip的依赖问题，请问该怎么解决

ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
nvidia-srl-usd 0.9.0 requires usd-core<24.00,>=21.11, which is not installed.
nvidia-srl-usd-to-urdf 0.3.2 requires usd-core<24.00,>=21.11, which is not installed.

Zhefan-Xu, 2025-03-22:
You can safely ignore those errors as this is not related to the training.

wei-02, 2025-03-22:
但是后面运行会报错，运行python training/script/train.py的时候，会报错无法打开training/script/train.py这个文件

Zhefan-Xu, 2025-03-22:
Could you please share your error?

wei-02, 2025-03-23:
运行完bash setup.sh后，会显示Setup completed successfully!
但是进入conda环境后，运行例程，报错

python: can't open file '/home/w/Desktop/NavRL/isaac-training/training/script/train.py': [Errno 2] No such file or directory

Zhefan-Xu, 2025-03-23:
You might want to check this issue: https://github.com/Zhefan-Xu/NavRL/issues/13#issue-2930919976

Zhefan-Xu, 2025-03-23:
Could you check whether this path is correct? /home/w/Desktop/NavRL/isaac-training/training/script/train.py

wei-02, 2025-03-24:
[image]
路径看着是对的，在这个路径下，是有对应的文件的

Zhefan-Xu, 2025-03-25:
Good to hear it works. It is suggested to use at least RTX 3080 and better to use a 4090 for training.