Zhefan-Xu/NavRL #16: Can RL be used for exploration?

paobuliao, 2025-03-22:
Reinforcement Learning (RL) has demonstrated significant potential in the field of navigation, prompting the question of whether it can be effectively applied to autonomous exploration. 
I am currently dedicating effort to designing a simulation environment and formulating a reward system. However, the calculation of rewards relies on past data, and the storage of mapped regions presents a challenge in implementation. I would greatly appreciate any insights you might have on this matter and look forward to hearing your opinion.

Zhefan-Xu, 2025-03-22:
Thanks for the question. I think RL can definitely be used in exploration if you set everything properly. For the specifc question regarding the past data, I would more think about incorporating past data in the states representation. But how to implement that is highly depends on your experiments. Also, I don't think it might be the only best option to train an E2E exploration and sometimes it could probably be better to seperate the exploration into submodules (exp goal selection + navigation) which might be easier to train.

Hope this can help.

paobuliao, 2025-03-23:
Thanks for the reply! I'm going to reward exploration only (exploration+navigation, as you mentioned). In a specific reward function setting, I might give each agent a separate memory for map recording and calculate the reward based on it. Hopefully everything goes well and the number of parallel training is not too low.
Thank you again for your help.

xiaowuguiOvO, 2025-04-11:
我也在尝试做RL探索，可以交流一下吗？