Zhefan-Xu/NavRL #18: when run in ROS1 without trian,how to modify setup.bash and orbit.sh to build a conda env NavRL？

traxu001, 2025-03-25:


traxu001, 2025-03-25:
I hope you can reply to me. Thank you very much

Zhefan-Xu, 2025-03-25:
Could you try save this as a ```setup_deployment.sh``` file under the same directory as the ```setup.sh```? Try running ```bash setup_deployment.sh```. Basically I just removed the things related to training, and please let me know if you have encounted any errors.

```
#!/bin/bash

# Exit immediately if a command fails
set -e

# Define environment name
ENV_NAME="NavRL"

# Load Conda environment handling
eval "$(conda shell.bash hook)"

# Step 1: Create conda env with python3.10
echo "Setting up conda env..."
conda create -n $ENV_NAME python=3.10
conda activate $ENV_NAME
pip install numpy==1.26.4
pip install "pydantic!=1.7,!=1.7.1,!=1.7.2,!=1.7.3,!=1.8,!=1.8.1,<2.0.0,>=1.6.2"
pip install imageio-ffmpeg==0.4.9
pip install moviepy==1.0.3
conda install pytorch==2.0.1 torchvision==0.15.2 torchaudio==2.0.2 pytorch-cuda=11.8 -c pytorch -c nvidia

# Step 2: Install TensorDict and dependencies
echo "Installing TensorDict dependencies..."
pip uninstall -y tensordict
pip uninstall -y tensordict
pip install tomli  # If missing 'tomli'
cd ../tensordict
python setup.py develop

# Step 3: Install TorchRL
echo "Installing TorchRL..."
cd ../rl
python setup.py develop

# Check which torch is being used
python -c "import torch; print(torch.__path__)"

echo "Setup completed successfully!"
```

traxu001, 2025-03-26:
Thank you very much for your help! Just changing line 29 to "cd third_party/tensordict" allowed me to install the conda env NavRL environment smoothly. But now I have a new problem. After opening three terminals in sequence and entering the commands, the robot model doesn't show up in Gazebo and RViz. Also, pressing the keyboard or setting navigation points has no effect. I'm a beginner, so I apologize for bothering you. I look forward to your help. The terminals are shown in the figure below:

roslaunch uav_simulator start.launch：

[image]
[image]

roslaunch navigation_runner safety_and_perception_sim.launch

[image]

conda activate NavRL
rosrun navigation_runner navigation_node.py

[image]

Zhefan-Xu, 2025-03-26:
Glad to hear that the env works and I will add the scripts to the repo so that people can directly deploy it without installing the training module.

For the model issue, have you put this line in your ```~/.bashrc```: 
```
echo 'source /path/to/ros1/uav_simulator/gazeboSetup.bash' >> ~/.bashrc
```
Make sure the path is all correct and then open a new terminal or run ```source ~/.bashrc```

traxu001, 2025-03-26:
I've added this command to my ~/.bashrc. Here's my .bashrc file: 

[image]

Are there any other changes or methods I can try ？  T-T

Zhefan-Xu, 2025-03-26:
Are you using virtual machine/WSL/docker container?

JinchuanMa, 2025-03-29:
> Are you using virtual machine/WSL/docker container?

Hi Dr. Xu, I'm having the exact same problem as him, I'm using WSL2, ubuntu 20.02 and ros noetic. i'm a beginner, sorry for bothering you, looking forward to your help.