Zhefan-Xu/NavRL #19: Run python training/scripts/train.py——ERROR

AdaptZZC, 2025-03-26:
Run python training/scripts/train.py，but it just have a window of isaac sim. This obstacle did not appear. 

And there is an error in the terminal. [Error] [rtx.neuraylib.plugin] [MDLC:COMPILER]   1.0   MDLC   comp error: http://omniverse-content-production.s3.us-west-2.amazonaws.com/Materials/vMaterials_2/Plastic/Plastic_Thick_Translucent.mdl?watch=000000003b0d0730(1,1): C100 "mdl" expected。

My CUDA Toolkit: 11.5, Driver: 12.4，GPU 4060, Driver Version: 550.120 

[image]

AdaptZZC, 2025-03-26:
[image]
Thank you for your reply!
After visiting this webpage, it looks like this. Does this mean that I may need to pre download some content？

Zhefan-Xu, 2025-03-26:
I am pretty sure that you encountered the internet issue. This causes an error when your isaac sim tried pulling some contents from Amazon US server. I never got similar issue but I heard people who got the issue pre download the things so that it can prevent that happens.