Zhefan-Xu/NavRL #21: The drone cannot be controlled via keyboard, rviz 2D navigation does not work, and obstacles do not move.

jiunian66, 2025-03-29:
Ubuntu20.04,Python 3.12.7,conda [24.9.2]

[image]

Zhefan-Xu, 2025-03-29:
Just want to check several things:
0. Is the drone hovering in Gazebo?
1. Are you able to control the drone by keyboard? There should be a control panel once you click it, you should be able to move it around. Just want to check  whether the controller is okay.
2. Are you using the latest Repo? The third terminal said dynamic obstacle func err should not appear in the latest version
3. Did you change any parameters?

jiunian66, 2025-03-29:
Thanks for you reply.We tried your first suggestion which the keyboard is already running,but the rviz 2D navigation does not work.The version is the latest and i didn't change any parameters

jiunian66, 2025-03-29:
[image]

Zhefan-Xu, 2025-03-29:
Any error on the second terminal? Except for the pcl error.

jiunian66, 2025-03-29:
Sorry, I sent the wrong picture，this picture is correct

[image]

Zhefan-Xu, 2025-03-29:
I am a bit confused. The second figure's error is different from the first one? Did you change anything and what is your current error

jiunian66, 2025-03-29:
We didn't change any parameters and now we have two errors
1：When publishing a target point using rviz 2D navigation this problem occurs, and the quadcopter stays at the origin in the simulation
(NavRL) jiu@jiu-System-Product-Name:~$ rosrun navigation_runner navigation_node.py
/home/jiu/anaconda3/envs/NavRL/lib/python3.10/site-packages/torch/nn/modules/lazy.py:180: UserWarning: Lazy modules are a new feature under heavy development so changes to the API or functionality can happen at any moment.
  warnings.warn('Lazy modules are a new feature under heavy development '
[nav-ros]: Press Enter to STOP motion!
[nav-ros]: take off completed at height:  1.0
Exception in thread Thread-24:
Traceback (most recent call last):
  File "/home/jiu/anaconda3/envs/NavRL/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/timer.py", line 237, in run
    self._callback(TimerEvent(last_expected, last_real, current_expected, current_real, last_duration))
  File "/home/jiu/path/to/catkin_ws/src/ros1/navigation_runner/scripts/navigation.py", line 239, in raycast_callback
    start_angle = np.arctan2(self.target_dir[1].cpu().numpy(), self.target_dir[0].cpu().numpy())
RecursionError: maximum recursion depth exceeded while calling a Python object
2：
[pcl::PCDReader::read] Could not find file 'No'.

jiunian66, 2025-03-31:
Whenever I use rziv 2D I get this error and the quadcopter just stays in place

[image]

Zhefan-Xu, 2025-03-31:
In the line causing the error (Line 239), can you break it down and see what portion of that line causes the error?

jiunian66, 2025-03-31:
[image]

[image]

Zhefan-Xu, 2025-03-31:
What I mean is that this line contains serval components such as ```np.arctan2```, ```.cpu()```, ```.numpy()``` operation and which component causes the issue.

Zhefan-Xu, 2025-04-01:
@last-venus Can you see whether the ros topic ```CERLAB/quadcoptor/odom```` exists or not?

last-venus, 2025-04-02:
WARNING: topic [/CERLAB/quadcoptor/odom] does not appear to be published yet

Zhefan-Xu, 2025-04-02:
> WARNING: topic [/CERLAB/quadcoptor/odom] does not appear to be published yet

You might have the Gazebo plugin issue. First verify you have included this line (with the correct path) to your ```~/.bashrc```
```
source /path/to/ros1/uav_simulator/gazeboSetup.bash
```

Sometimes, virtual machine uses a different path which you might need to check.

1824243230, 2025-04-02:
[image] home/hyh/anaconda3/envs/NavRL/lib/python3.10/site-packages/torch/nn/modules/lazy.py:180: UserWarning: Lazy modules are a new feature under heavy development so changes to the API or functionality can happen at any moment.
  warnings.warn('Lazy modules are a new feature under heavy development '
[nav-ros]: Press Enter to STOP motion!
[nav-ros]: take off completed at height:  1.0
Exception in thread Thread-24:
Traceback (most recent call last):
  File "/home/hyh/anaconda3/envs/NavRL/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/timer.py", line 237, in run
    self._callback(TimerEvent(last_expected, last_real, current_expected, current_real, last_duration))
  File "/home/hyh/NavRL_ws/src/ros1/navigation_runner/scripts/navigation.py", line 239, in raycast_callback
    start_angle = np.arctan2(self.target_dir[1].cpu().numpy(), self.target_dir[0].cpu().numpy())
RecursionError: maximum recursion depth exceeded while calling a Python object  
Hello Dr. Xu, I have fixed the issue with loading the Gazebo model. The navigation module is experiencing this problem, rviz 2D navigation does not work.

Zhefan-Xu, 2025-04-02:
@last-venus @jiunian66 @1824243230 

I tested with the ```bash setup_deployment.sh``` myself and I reproduce the error you met. It seems that there are bugs in the ```setup_deployment.sh``` file and I had already fixed that in the latest version. 

To address the issue you met, first remove the NavRL conda environment:
```
conda remove --name NavRL --all
```
Then, download the latest version and rerun ```bash setup_deployment.sh```. After that, you can follow the same steps mentioned in this repo.

I have verified that the current version should work and please let me know if you still face any issue! Thanks

jiunian66, 2025-04-03:
Thank you very much for your help! The project can run normally and is very stable .The simulation performed well!!!

[image]

Zhefan-Xu, 2025-04-03:
@jiunian66 Thank you very much for the feedback. I think you might also want to update some default paramters that I changed as well by pulling the latest commit.

jiunian66, 2025-04-03:
I have used the latest commit，thanks a lot!

1824243230, 2025-04-03:
Thank you Dr. Xu for your busy schedule to reply to the bug fixes, I hope you have good health and all the best!

HYLCXH, 2025-06-25:
> Thank you very much for your help! The project can run normally and is very stable .The simulation performed well!!!
> 
> [image]

My obstacles do not move as well,how do you slove the problm