Zhefan-Xu/NavRL #23: How to set the reward

Bruce-Si, 2025-03-31:
Thank you for your great work. I have a question about the reward function：
What does the +1. represent in `self.reward`?
Also, it seems that  the update of `self.prev_drone_vel_w` doesn't work, so it always equals to `self.drone.vel_w[..., :3]`.

Zhefan-Xu, 2025-03-31:
+1 is the reward that encourage the robot to survive.

```self.prev_drone_vel_w``` is to store the velocity from the previous time step. ```self.drone.vel_w[..., :3]``` will be updated by the Isaac Sim.

Bruce-Si, 2025-03-31:
But in my case, it's strange that `self.drone.vel_w[..., :3]` gets updated too early — as a result, `penalty_smooth` is always zero.

Zhefan-Xu, 2025-03-31:
I am not sure what do you mean by "gets updated too early". Did you mean that you print out ```penalty_smooth``` and all the values are zero (at some late training stage).

Zhefan-Xu, 2025-03-31:
I had just tested myself and I think you are correct. The issue is that whenever ```self.drone.vel_w``` is updated by the simulator, the previous velocity ```self.drone.prev_vel_w``` is also updated (since it is assigned by reference). This is a mistake that we made during open sourcing and it is not correct by the commit here https://github.com/Zhefan-Xu/NavRL/commit/1f689e5cc4637610762d546052ec3d84630bbc0e.

The things that I changed for ```self.drone.prev_vel_w``` is to let it clone the tensor instead of assigning the value by reference.

Thank you very much for pointing this out! If you still observe an issue, please do let me know!

Bruce-Si, 2025-04-01:
Thank you for your reply.
When `env_dyn.num_obstacles` is set to zero, the code proceeds to `dyn_obs_states = torch.zeros(self.num_envs, 1, self.cfg.algo.feature_extractor.dyn_obs_num, 8, device=self.cfg.device)`.
The value 8 might need to be changed to 10, as it represents `num_dim_each_dyn_obs_state`.

Zhefan-Xu, 2025-04-01:
> Thank you for your reply. When `env_dyn.num_obstacles` is set to zero, the code proceeds to `dyn_obs_states = torch.zeros(self.num_envs, 1, self.cfg.algo.feature_extractor.dyn_obs_num, 8, [...]

Thanks for catching that up and I had corrected it.

closed 2025-04-07 by Bruce-Si