Zhefan-Xu/NavRL #24: whem I moved the download isaas sim from the docker container to local machine,I have encountered a problem

wms848, 2025-04-01:
[image]

wms848, 2025-04-01:
When I input bash docker ps,screen output error code

Zhefan-Xu, 2025-04-01:
You have to do that in a seperate terminal (outside the docker)

wms848, 2025-04-01:
thanks for your help,now I meet a new question

[image]

wms848, 2025-04-01:
I set the ISAACSIM_PATH environment variable

[image]

Zhefan-Xu, 2025-04-02:
> thanks for your help,now I meet a new question
> 
> [image]

From here, you should set ```path/to/Isaac...``` to the actual path

wms848, 2025-04-02:
How do you set the actual path path/to/Isaac...?

Zhefan-Xu, 2025-04-02:
For example ```~/home/Desktop/Isaac...``` instead of ```path/to/Isaac```

wms848, 2025-04-03:
I would like to know where to modify this actual path

wms848, 2025-04-03:
I want to know which file I need to modify.