Zhefan-Xu/NavRL #25: Can I run the ros1 program in CPU?

JulienZHENG, 2025-04-02:
I am trying to deploy this work in a drone with CPU and I am wondering about its feasibility.

JulienZHENG, 2025-04-02:
It is possible. I have installed the cpu version of pytorch==2.4.0 and modified the related config. The simulation performed well. I'll keep trying in a real drone device

Zhefan-Xu, 2025-04-02:
Yes, it is available to run on CPU since the model is relatively small. If you try running on the real drone, make sure it has enough CPU computation power

Mming02, 2025-04-07:
hello, Dr Xu, when the ros1 program run only with cpu, i can use the rviz 2D Nav Goal to guide the drone move. But why in the gazebo there are only static obstacles and no dynamic obstacles?

[image]

[image]

Zhefan-Xu, 2025-04-07:
@Mming02 Thanks for the comment. As the repo mentioned, you can change the environment by modifying the ```start.launch``` file and there we provided a static, purely dynamic, and a mixed environment.