Zhefan-Xu/NavRL #26: bash setup.sh ->The read operation timed out

wangchengwei-wcw, 2025-04-02:
[image]

[image]

运行bash setup.sh与bash setup_deployment.sh都会出现The read operation timed out的问题，翻墙与不翻墙都尝试了，还是会timed out。后面尝试自己手动下载。手动下载成功了，但是在运行上述两行指令时，还是会自动检测刚才手动下载的内容，然后继续timed out。不知如何解决。

Zhefan-Xu, 2025-04-03:
You can try manually run the command in the file step by step. If you re-run the entire script, it will create a new conda environment and your manually downloaded packages will not be used.

w4646, 2025-04-12:
> You can try manually run the command in the file step by step. If you re-run the entire script, it will create a new conda environment and your manually downloaded packages will not be used.

Hello, when I run this command: bash setup. sh, I get an error as shown in the picture. What's going on

[image]

Zhefan-Xu, 2025-04-12:
@w4646 Are you using the latest version? Perhaps update to the latest version and then try remove the current conda env by ```conda remove --name NavRL --all``` and rerun the ```bash setup.sh```.

Ysysqjr, 2025-04-29:
@wangchengwei-wcw 想请问下您这个问题最后如何解决的呢？我现在也遇到了time out的问题