Zhefan-Xu/NavRL #29: Deployed on melodic

wei-02, 2025-04-03:
Can this be deployed on melodic? I deployed it on melodic and got an error when starting the navigation：

[image]

Referring to the csdn article, the solution given is to compile the tf2 package with python3, and then change the path to my path. The link is as follows:

https://blog.csdn.net/xiaobaijinhuaji/article/details/128226957

But I can't find where to import the tf2 package. Is there any solution?

Zhefan-Xu, 2025-04-03:
1. This package can be used in ROS melodic.
2. There might be some changes in tf package from melodic to noetic. These functions are mainly used for coordinate transform. So my suggestion is:
 - Find the correponding ros tf package in melodic and made modification to this pkg accordingly.
 - Or, use any coordinate transform package to perform the task.

Both ways should not be complicated.

wei-02, 2025-04-14:
I have solved the problem last time. The reason why I failed to solve it before was because I didn’t source it in the terminal. But now I have a new problem. I can’t set the target for navigation. This is the screen after startup:
[image]

However, when setting the 2D Navl goal in rviz, the navigation terminal will exit, as shown in the following figure:
[image]

I don't know what the cause is, and when I start rviz, the terminal will report an error, it seems that the point cloud map cannot be found. Does this error have any impact?
[image]

Zhefan-Xu, 2025-04-14:
Thanks for the feedback. Are you using 18.04? If that is the case, could you please chase down which line causes the issue and I can try figuring it out? 

Also, perhaps this issue is related and it might be due to an incorret installation of cuda driver: https://github.com/moveit/moveit/issues/2234#issue-672009646

wei-02, 2025-04-16:
Yes, I am using 18.04. Regarding the error that the point cloud map cannot be found, I found that in the four yaml files for loading the map, the parameter prebuilt_map_directory has "No" in some files and your path in others. I changed them all to my corresponding paths, as shown below:

[image]

When I run the code again, the error reported by the terminal about pcl is gone, but the drone still does not move when the target is given in rviz, and the terminal will report an error, but it will not crash.

[image]

[image]

Although the problem of not finding the point cloud file has been solved, I think the drone should build the map while exploring, so I changed the parameter value of prebuilt_map_directory to "No", but the previous problem still occurs. When starting the rviz interface, the point cloud map file cannot be reported, and when setting a target for the drone, it still crashes as soon as the target is set:

[image]

Is there any connection between these two questions?Moreover, regardless of whether there is a point cloud map or not, the drone will not move towards the target point I set.

Zhefan-Xu, 2025-04-16:
The new error is fixed here https://github.com/Zhefan-Xu/NavRL/issues/21#issuecomment-2773738551

Perhaps you are not using the latest version.