Zhefan-Xu/NavRL #31: docker The package cannot be imported during training

xch1176411706, 2025-04-06:
Hello, may I ask if you have encountered this problem again? Several packages are missing separately. How to solve it

[image]

[image]

Zhefan-Xu, 2025-04-06:
Thanks for posting the issue. Could you show what error you get? Those IDE underlines should not be a problem.

xch1176411706, 2025-04-08:
感谢您的回复，可以直接运行training/scripts/train.py,我在没有桌面的服务器运行的时候，headless: True设置，出现导入环境的问题，请问没有桌面怎么解决

[image]

xch1176411706, 2025-04-08:
Headless mode: True
[150.039s] [ext: omni.physx.fabric-105.1.10-5.1] startup
2025-04-08 02:42:33 [150,071ms] [Warning] [omni.fabric.plugin] No source has valid data array=0x3dddb350 usdValid=0 cpuValid=0
[2025-04-08 10:42:33,075][root][INFO] - Reading Hummingbird's params from /home/xuech23/NavRL/isaac-training/third_party/OmniDrones/omni_drones/robots/assets/usd/hummingbird.yaml.
2025-04-08 02:45:06 [303,505ms] [Warning] [omni.usd] Warning: in _ReportErrors at line 2890 of /buildAgent/work/ac88d7d902b57417/USD/pxr/usd/usd/stage.cpp -- In </World/defaultGroundPlane>: Could not open asset @http://omniverse-content-production.s3-us-west-2.amazonaws.com/Assets/Isaac/2023.1.0/Isaac/Environments/Grid/default_environment.usd@ for reference introduced by @anon:0x3dc6ca30:World0.usd@</World/defaultGroundPlane>. (recomposing stage on stage @anon:0x3dc6ca30:World0.usd@ <0x3dd1b5d0>)

Error executing job with overrides: []
Traceback (most recent call last):
  File "/home/xuech23/NavRL/isaac-training/training/scripts/train.py", line 49, in main
    env = NavigationEnv(cfg)
  File "/home/xuech23/NavRL/isaac-training/third_party/rl/torchrl/envs/common.py", line 159, in __call__
    instance: EnvBase = super().__call__(*args, **kwargs)
  File "/home/xuech23/NavRL/isaac-training/training/scripts/env.py", line 40, in __init__
    super().__init__(cfg, cfg.headless)
  File "/home/xuech23/NavRL/isaac-training/third_party/OmniDrones/omni_drones/envs/isaac_env.py", line 134, in __init__
    global_prim_paths = self._design_scene() # in the child class, init robot (1), light, terrain (with number of env)
  File "/home/xuech23/NavRL/isaac-training/training/scripts/env.py", line 101, in _design_scene
    cfg_ground.func("/World/defaultGroundPlane", cfg_ground, translation=(0, 0, 0.01))
  File "/home/xuech23/NavRL/isaac-training/third_party/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/sim/spawners/from_files/from_files.py", line 145, in spawn_ground_plane
    collision_prim_path = prim_utils.get_prim_path(
  File "/home/xuech23/isaac-sim/exts/omni.isaac.core/omni/isaac/core/utils/prims.py", line 349, in get_prim_path
    return prim.GetPath()
AttributeError: 'NoneType' object has no attribute 'GetPath'

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.
wandb: Waiting for W&B process to finish... (failed 1).