Zhefan-Xu/NavRL #33: About the map in Rviz during 'NavRL ROS1 Deployment'

Metaph0r715, 2025-04-07:
# Start the perception and safety module
roslaunch navigation_runner safety_and_perception_sim.launch

When I tried to start the Rviz in this step, there is no map in Rviz. 
Error is showed as following pictures.
[image]
[image]
I guess it may be caused by the wrong path to pcd, but I don't know how to solve it.
Thanks for your any help!

Zhefan-Xu, 2025-04-07:
Thanks for posting the issue. I noticed that there is no drone in the simulator. I would suggest to check whether your ```~/.bashrc``` contains this line ```source /path/to/ros1/uav_simulator/gazeboSetup.bash```

Metaph0r715, 2025-04-08:
> Thanks for posting the issue. I noticed that there is no drone in the simulator. I would suggest to check whether your `~/.bashrc` contains this line `source /path/to/ros1/uav_simulator/gazeboSetup.bash`

Thanks for your reply! About the `~/.bashrc` contains, I had added that line before this error happened, and you can see it in `line 148` in the following picture.
[image]
Besides, it had reported the wrong path in the terminal before I set the absolute path, so I guess it works successfully?

Zhefan-Xu, 2025-04-08:
You should deactivate the conda env for all other ROS launch except for the navigation_node

Metaph0r715, 2025-04-08:
> You should deactivate the conda env for all other ROS launch except for the navigation_node

OK, I have tried to execute `conda deactivate` on the terminal before `roslaunch uav_simulator start.launch` just now, and it works successfully. Thank you so much!

[image]

closed 2025-04-08 by Metaph0r715