Zhefan-Xu/NavRL #35: Problems during the Ros1 compilation process

Metaph0r715, 2025-04-08:
I successfully compiled and ran the joint simulation of Gazebo and RVIZ yesterday, but when I tried `deployed NavRL ROS1` again today, there were  many error messages, as shown in the following figure.
[image]
I followed the AI prompts to reinstall Gazebo and switch to a specific version of Protobuf, but there were no changes at all.
Thanks for any help！

Zhefan-Xu, 2025-04-08:
This is more like a Gazebo version issue. I would suggest to fix your ROS gazebo installation.

Metaph0r715, 2025-04-08:
> This is more like a Gazebo version issue. I would suggest to fix your ROS gazebo installation.

Thanks so much for your reply!
I have completely reinstalled ROS and Gazebo, and the Gazebo interface can also be opened normally. 
There is no problem compiling the other three packages in ros1, only when compiling the `uav_simulator` will there be such errors consistently. 
After successfully compiling and running yesterday, I downloaded and compiled another project containing `uav_simulator`. I suspect that some version was modified during the process. Is `protoc 3.6.1` normal?

Zhefan-Xu, 2025-04-09:
I would suggest trying to install the simulator in a clean ros gazebo environment: https://github.com/Zhefan-Xu/uav_simulator

If any compilation error, then please post it.

lmhqsy, 2025-04-12:
> I would suggest trying to install the simulator in a clean ros gazebo environment: https://github.com/Zhefan-Xu/uav_simulator
> 
> If any compilation error, then please post it.

> This is more like a Gazebo version issue. I would suggest to fix your ROS gazebo installation.
I reinstalled ROS and Gazebo, and also created a brand-new workspace. In the `src` directory, I only cloned the `uav_simulator` package. However, when I compile, the same error still occurs. What should I do now?

Xinyu928-hub, 2025-04-13:
I'm encountering the same issue—`Invoking "make -j16 -l16" failed.` What's puzzling is that I didn't have any problems with `catkin_make` for `map_manager` and `onboard_detector`, but the issue arose after adding `uav_simulator`. I'm using Ubuntu 20.04, ROS 1, and Gazebo 11. I'm not sure how to resolve this and would appreciate your response, especially with more detailed version information.

lmhqsy, 2025-04-13:
>

I deleted everything related to `protoc` in Ubuntu and reinstalled it, which solved the problem.

Doc-Lbb, 2025-04-20:
> > > 我建议尝试在干净的 ros gazebo 环境中安装模拟器：https://github.com/Zhefan-Xu/uav_simulator 如果有任何编译错误，请发布它。
> > 
> > 
> > > 这更像是 Gazebo 版本问题。我建议修复您的 ROS 凉亭安装。我重新安装了 ROS 和 Gazebo，还创建了一个全新的工作区。在该目录中，我只克隆了包。但是，当我编译时，同样的错误仍然会出现。我现在该怎么办？`src``uav_simulator`
> 
> 我删除了 Ubuntu 中的所有内容并重新安装它，这解决了问题。`protoc`

请问是重装了ubuntu系统吗，还是只是把ROS相关的删除了