Zhefan-Xu/NavRL #36: Issue with Gazebo and RViz Display(ROS1)

nhhxx0018, 2025-04-09:
Hi Dr.Xu, thank you so much for making this project open source:)
I'm using virtual machine(ubuntu 20.04 and ros noetic). I have followed the instructions in the README file, and after opening three terminals in sequence and entering the provided commands, I encountered an issue where the robot model does not show up in Gazebo or RViz. Additionally, pressing the keyboard or setting navigation points has no effect.（same as issue 18）

I check the points you made in issue 18, but simulation still appears blank during the generation phase now.like this:

[image]

[image]

[p.s.]
1.In the "Training in Isaac Sim" section, it says that the training part can be skipped, so I did not train the model and continued with the reproduction process. Could this be related to the blank screen issue?

2.This file appears in compressed form in the virtual machine. Do I need to unzip it? [https://github.com/Zhefan-Xu/NavRL/blob/main/ros1/navigation_runner/scripts/ckpts/navrl_checkpoint.pt]
As a beginner, I apologize for any inconvenience, and I would appreciate your guidance on this matter.

Thank you for your time and assistance. I look forward to your response.

Zhefan-Xu, 2025-04-09:
I think everything else should be fine except that you are using WSL. Some people told me WSL needs to set the Gazebo environment variable differently.

I checked online and perhaps this can help: https://www.reddit.com/r/ROS/comments/11wh0m3/wsl2_does_not_display_gazebo_and_rviz/?rdt=37741

nhhxx0018, 2025-04-11:
thx for your reply. Seems awkward cuz I check this sentence and find I set wrong in conda initialize. The **path/to** should refer to **catkin_ws/src**  instead of **NavRL** :)
echo 'source /path/to/ros1/uav_simulator/gazeboSetup.bash' >> ~/.bashrc