Zhefan-Xu/NavRL #39: PhysX GPU buffer settings from sim.yaml not applied before environment initialization

Hammerandsickle, 2025-04-13:
I'm trying to run multi-UAV parallel training on an RTX 3080 using the NavRL project. I configured PhysX GPU buffer parameters (e.g., totalAggregatePairsCapacity) in sim.yaml, but the simulator still seems to use the default value 1024 at runtime.

As a result, I encounter the following error during training: `PhysX error: The application needs to increase PxgDynamicsMemoryConfig::totalAggregatePairsCapacity to XXXX
`
**Environment**

    GPU: NVIDIA RTX 3080

    Isaac Sim: 2023.1.0-hotfix.1

    Training config: env.num_envs=60 env.num_obstacles=150 env_dyn.num_obstacles=8 

**Request**

Could you please confirm:

1.     What is the recommended way to ensure sim.yaml PhysX configs are applied correctly?

2.     Should the default totalAggregatePairsCapacity be overwritten manually in env.py?

Any guidance would be greatly appreciated!

[image]

[image]

Zhefan-Xu, 2025-04-13:
Thanks for posting the issue. Did you have issue with training 2 drones? Would like to confirm that it is not installation issue.

Hammerandsickle, 2025-04-14:
Hi, thanks for the quick response!

I can confirm that training works fine with up to 50 drones.
However, when I increase the number of drones beyond that, I encounter the following error:

`PhysX error: The application needs to increase PxgDynamicsMemoryConfig::totalAggregatePairsCapacity ...
`
I also checked the GPU PhysX configuration panel in Isaac Sim, and found that the totalAggregatePairsCapacity is set to 1024, which I believe is the default value. This seems to be the root cause of the error when scaling to more drones.

Let me know if there's a recommended way to configure this properly via config or code—currently I’m testing a few approaches.

Thanks again!

Zhefan-Xu, 2025-04-14:
The RTX 3080 can only train up to 256 drones with ```headless=True``` based on my experience since it only has 10GB memory.

Hammerandsickle, 2025-04-14:
Thanks for the clarification!

In my case, I'm getting the totalAggregatePairsCapacity error even with just 60 drones on an RTX 3080 (10GB).
I checked the PhysX GPU panel in Isaac Sim, and the `totalAggregatePairsCapacity` is still at the default value of 1024.

Although I’ve already set this parameter to a larger value (e.g., 20000) in sim.yaml, it seems the value didn't get passed into the simulation.
I’m not sure how to properly modify the default PhysX configuration or ensure that the settings from sim.yaml are actually applied.

Would appreciate any advice on how to fix this properly or where the right entry point is to make sure the sim config is respected.

Thanks again!

Zhefan-Xu, 2025-04-14:
That's weird...but 20000 will defintely not work as the bottleneck will not be the GPU memory then. Have you tried ```headless=True```?

Hammerandsickle, 2025-04-14:
Yes, I’m using headless=True, and here’s some clarification:

I didn’t actually set the totalAggregatePairsCapacity to 20000 — I set it to 2097152, along with the other PhysX GPU buffer parameters (screenshot below).

[image]

Despite that, I still encounter the error when increasing to just 60 drones. The following error is `PhysX error: The application needs to increase PxgDynamicsMemoryConfig::totalAggregatePairsCapacity to 1034
`

It seems like the config is either not being applied at runtime or being overwritten somewhere, since the value shown in the GPU PhysX debug panel remains at the default 1024.

[image]