Zhefan-Xu/NavRL #40: bash setup.sh

Vigoss-z, 2025-04-13:
Hello Dr. Xu, I encountered this error when executing bash setup.sh:

`Setting up Orbit...
Running orbit.sh setup...
[INFO] Using conda environment name: NavRL                                     
[INFO] Conda environment named 'NavRL' already exists.
/home/dell/anaconda3/envs/NavRL/etc/conda/activate.d/setenv.sh: 行 4: /home/dell/NavRL-main/isaac-training/third_party/orbit/_isaac_sim/setup_conda_env.sh: 没有那个文件或目录`

I followed the installation steps.
docker cp bb62939be6d3:isaac-sim/. /home/dell/isaac:
[image]
echo 'export ISAACSIM_PATH="home/dell/isaac"' >> ~/.bashrc:
[image]

How can I resolve this?

Zhefan-Xu, 2025-04-13:
Try putting ```export ISAACSIM_PATH="/home/dell/isaac/isaac_sim-2023.1.0-hotfix.1``` to your ```~/.bashrc```. Then re-run ```bash setup.bash```

Vigoss-z, 2025-04-13:
> Try putting `export ISAACSIM_PATH="/home/dell/isaac/isaac_sim-2023.1.0-hotfix.1` to your `~/.bashrc`. Then re-run `bash setup.bash`

After trying as you said:

[image]

Zhefan-Xu, 2025-04-13:
Please remove the environment first by ```conda remove --name NavRL --all```. Then redo the procedures and please let me know if the issue still persists.

Vigoss-z, 2025-04-13:
> Please remove the environment first by `conda remove --name NavRL --all`. Then redo the procedures and please let me know if the issue still persists.
the issue still persists:
[image]
[image]

Vigoss-z, 2025-04-13:
Is it related to the NVIDIA driver version? I installed 550.78

Zhefan-Xu, 2025-04-13:
I don't think the issue comes from the driver version. It seems the error is due to the incorrect creation of the ```ISAACSIM_PATH``` symbolic link.

Could you try doing this: 
```
cd path/to/NavRL/isaac-training/third_party/orbit

ln -s ${ISAACSIM_PATH} _isaac_sim
```

Let me know if that throws an error. If everything works fine, then you will see a ```_isaac_sim``` folder.

Vigoss-z, 2025-04-14:
> I don't think the issue comes from the driver version. It seems the error is due to the incorrect creation of the `ISAACSIM_PATH` symbolic link.
>
> Could you try doing this:
>
> ```
> cd [...]

I see a `_isaac_sim` folder.But when I run it, it shows:
[image]
the issue still persists when executing bash setup.sh.

IsabeldeCastilla, 2025-04-14:
> I see a `_isaac_sim` folder.But when I run it, it shows: [image] the issue still persists when executing bash setup.sh.

Hi, could you try to first do ```conda remove --name NavRL --all```

Then please change the folder name under ```/home/dell``` from ```isaac``` to ```isaac_sim-2023.1.0-hotfix.1```, which makes the whole path ```/home/dell/isaac_sim-2023.1.0-hotfix.1```.

Then try put ```export ISAACSIM_PATH="/home/dell/isaac/isaac_sim-2023.1.0-hotfix.1"``` to your ~/.bashrc. Then re-run ```bash setup.bash``` and try again?

IsabeldeCastilla, 2025-04-14:
> I see a `_isaac_sim` folder.But when I run it, it shows: [image] the issue still persists when executing bash setup.sh.

Hi, could you please first do ```conda remove --name NavRL --all```

Then please change the folder name under ```/home/dell``` from ```isaac``` to ```isaac_sim-2023.1.0-hotfix.1```, which makes the whole path ```/home/dell/isaac_sim-2023.1.0-hotfix.1```

Then try put ```export ISAACSIM_PATH="/home/dell/isaac_sim-2023.1.0-hotfix.1"``` to your ~/.bashrc. Then re-run ```bash setup.bash``` and try again?

Vigoss-z, 2025-04-14:
> > > I don't think the issue comes from the driver version. It seems the error is due to the incorrect creation of the `ISAACSIM_PATH` symbolic link.
> > > Could you try doing this:
> > > ```
> > > cd path/to/NavRL/isaac-training/third_party/orbit
> > > 
> > > ln -s ${ISAACSIM_PATH} _isaac_sim
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > Let me know if that throws an error. If everything works fine, then you will see a `_isaac_sim` folder.
> > 
> > 
> > I see a `_isaac_sim` folder.But when I run it, it shows: [image] the issue still persists when executing bash setup.sh.
> 
> Hi, could you try to first do `conda remove --name NavRL --all`
> 
> Then please change the folder name under `/home/dell` from `isaac` to `isaac_sim-2023.1.0-hotfix.1`, which makes the whole path `/home/dell/isaac_sim-2023.1.0-hotfix.1`.
> 
> Then try put `export ISAACSIM_PATH="/home/dell/isaac_sim-2023.1.0-hotfix.1"` to your ~/.bashrc. Then re-run `bash setup.bash` and try again?

I tried what you said,the issue still persists.

IsabeldeCastilla, 2025-04-14:
> I tried what you said,the issue still persists.

Thank you for the reply! could you do this again and let me know is there still an issue for this symbolic link and what the issue is?
```
cd path/to/NavRL/isaac-training/third_party/orbit

ln -s ${ISAACSIM_PATH} _isaac_sim
```

Vigoss-z, 2025-04-14:
@IsabeldeCastilla I do it again,same as before

IsabeldeCastilla, 2025-04-14:
> [@IsabeldeCastilla](https://github.com/IsabeldeCastilla) I do it again,same as before

Could you please double check the path in your ~/.bashrc? It should be ```export ISAACSIM_PATH="/home/dell/isaac_sim-2023.1.0-hotfix.1"```, but not ```export ISAACSIM_PATH="/home/dell/isaac/isaac_sim-2023.1.0-hotfix.1"```. 

The symbolic link should work as long as it points to an existing folder, so please make sure the directory exists. Just to confirm: this time the path should be directly under /home/dell/, without the extra isaac/ in between.
Thank you.

Vigoss-z, 2025-04-14:
> Could you please double check the path in your ~/.bashrc? It should be `export ISAACSIM_PATH="/home/dell/isaac_sim-2023.1.0-hotfix.1"`, but not `export [...]

Thank you very much.I understand what you mean. I have tried everything you said,the issue still persists.I also reinstalled isaac_sim-2023.1.0-hotfix.1 before, but the issue still persists.

Zhefan-Xu, 2025-04-14:
> I see a `_isaac_sim` folder.But when I run it, it shows: [image] the issue still persists when executing bash setup.sh.

```home/``` is not a valid path but instead you will have to type ```/home``` (Don't omit ```/```). You can easily verify the validity of your path this by ```cd /home/...(your path)```

Vigoss-z, 2025-04-14:
Thank you for your reply, it has wasted your time I have already resolved it

closed 2025-04-14 by Vigoss-z