Zhefan-Xu/NavRL #44: which torch version is correct, 2.0.1 or 2.1.0?

KurtRao, 2025-04-19:
If I install torch==2.1.0, it says:
```
omni-isaac-orbit-tasks 0.6.0 requires torch==2.0.1, but you have torch 2.1.0 which is incompatible.
omni-isaac-orbit 0.15.1 requires torch==2.0.1, but you have torch 2.1.0 which is incompatible.
```

however, if I install torch=2.0.1, it says:
```
tensordict 0.4.0+df5b74a requires torch>=2.1.0, but you have torch 2.0.1 which is incompatible.
torchrl 0.4.0+df5b74a requires torch>=2.1.0, but you have torch 2.0.1 which is incompatible.
```

Zhefan-Xu, 2025-04-19:
Please use the ```setup.bash``` (training + deployment) or ```setup_deployment.sh``` (deployment only) to install the virtual environment. You don't have to install ```torch``` manually and the version that we installed is ```2.0.1```.

KurtRao, 2025-04-20:
Understand. Thank you. It does work correctly under 2.0.1

closed 2025-04-20 by KurtRao