Zhefan-Xu/NavRL #47: ros1无人机无法启动

Doc-Lbb, 2025-04-21:
请问我在nav-ros之后，无人机一直停留在 Wait for robot odom...，无人机也是一直在地面，没有启动,然后操作手柄也点不动，检查环境变量也正常
系统ubuntu20.04 

[image]

AdoreMoney, 2025-04-21:
请问一下你启动navigation_node的时候有碰到ModuleNotFoundError: No module named 'rospkg'这个问题吗  我查了一下说是NavRL环境是3.10的 但是系统apt的rospkg只兼容python3.8

Doc-Lbb, 2025-04-21:
> 请问一下你启动navigation_node的时候有碰到ModuleNotFoundError： No module named 'rospkg'这个问题吗 我查了一下说是NavRL环境是3.10的 但是系统apt的rospkg只兼容python3.8

没有遇到这个错误，在运行nav_ros之后都没有错误，就一直等待

Zhefan-Xu, 2025-04-21:
Thanks for posting the issue. Can you check your rostopic?

Specifically ```/CERLAB/quadcopter/odom```. If you don't see data in this topic, it means that your environment variable is not properlly set.

Doc-Lbb, 2025-04-22:
> 感谢您发布此问题。您可以检查一下您的 rostopic 吗？
> 
> 具体来说`/CERLAB/quadcopter/odom`，如果您在此主题中没有看到数据，则意味着您的环境变量未正确设置。

[image],查看了一下，话题是有的

Zhefan-Xu, 2025-04-22:
@Doc-Lbb Thanks for the feedback.

Could you please check the data by ```rostopic echo /CERLAB/quadcopter/odom```

Doc-Lbb, 2025-04-23:
@Zhefan-Xu 谢谢您的建议
我运行了rostopic echo /CERLAB/quadcopter/odom
出现：WARNING: no messages received and simulated time is active.
Is /clock being published?
我检查roscore也是打开的，并且我运行rostopic info /CERLAB/quadcopter/odom
出现如下图

[image]

Zhefan-Xu, 2025-04-23:
Could you please show your bashrc file? Just want to check whether the environment variable is set correctly. Also, are you using virtual machine?

Doc-Lbb, 2025-04-23:
> 你能展示一下你的 bashrc 文件吗？我只是想检查一下环境变量是否设置正确。另外，你使用的是虚拟机吗？

[image]
我用的不是虚拟机，用的实体机

Zhefan-Xu, 2025-04-23:
this path is supposed to be under the catkin workspace (the directory after you copying ros1 into your workspace)

Doc-Lbb, 2025-04-23:
@Zhefan-Xu 谢谢提醒，已经修复该错误，现在运行之后，nav-ros停留在如下图，请问正常吗

[image]

Zhefan-Xu, 2025-04-23:
Looks alright. If you can see the perception range of the robot, you can play with it by clicking the Rviz ```2D Nav goal``` to select a goal position. Make sure your goal position is not within the obstacle.

Doc-Lbb, 2025-04-23:
> 看起来还不错。如果你能看到机器人的感知范围，就可以点击 Rviz`2D Nav goal`选择目标位置来尝试一下。确保目标位置不在障碍物范围内。

[image]
效果很好，谢谢徐博士耐心解答！

ShuibeiShannan, 2025-05-17:
> 请问我在nav-ros之后，无人机一直停留在 Wait for robot odom...，无人机也是一直在地面，没有启动,然后操作手柄也点不动，检查环境变量也正常 系统ubuntu20.04
> 
> [image]

你好，打扰啦，请问下gazebo其中的无人机模型需要额外加载嘛，我启动后没有这个模型

[image]