Zhefan-Xu/NavRL #50: The drone cannot be controlled

yangxueyan0910, 2025-04-21:
Thanks to the author for open source,I like your work very much！

My operating environment is: wsl2, ubuntu20.04, ros1.
After starting the simulation and deploying NavRL navigation according to the readme, the drone can only fly for a few seconds and then cannot be controlled. As shown in the terminal below: "take off completed at height: 1.0", and then the drone can no longer be controlled.

I run the following two lines of commands outside the virtual environment NavRL
"roslaunch uav_simulator start.launch"
"roslaunch navigation_runner safety_and_perception_sim.launch"

[image]

[image]

[image]

yangxueyan0910, 2025-04-21:
I have already fixed this problem!

Zhefan-Xu, 2025-04-21:
Thanks for posting the issue. Good to know that the issue was addressed. If you could share how you resolve this problem, it would be very helpful to people who had similar issues. Thanks!