Zhefan-Xu/NavRL #51: 安装orbit的时候，出现thirty_party/orbit/低下没有_isaac_sim/setup_conda_env.sh的情况

charkyli, 2025-04-22:
请问博主有遇到这种情况吗，这是哪一步没有设置好呢，已经把Isaac sim拷贝到本地路径下了
Setting up Orbit...
Removing existing symbolic link: _isaac_sim
Running orbit.sh setup...
[INFO] Using conda environment name: NavRL                                                                                                                                            
[INFO] Conda environment named 'NavRL' already exists.
/home/lcq/miniconda3/envs/NavRL/etc/conda/activate.d/setenv.sh: line 4: /home/lcq/projects/NavRL/isaac-training/third_party/orbit/_isaac_sim/setup_conda_env.sh: No such file or directory

Zhefan-Xu, 2025-04-22:
Please check this: https://github.com/Zhefan-Xu/NavRL/issues/40#issue-2990999287

Zhefan-Xu, 2025-04-22:
Thanks for the feedback.

Could you please check whether you have ```setup_conda_env.sh``` under ```/home/lcq/isaac-sim/isaac_sim-2023.1.0-hotfix.1```?

charkyli, 2025-04-22:
> 感谢您的反馈。
> 
> 您能否检查一下您是否有 ？`setup_conda_env.sh``/home/lcq/isaac-sim/isaac_sim-2023.1.0-hotfix.1`

[image]
这个/home/lcq/isaac-sim/路径下是有setup_conda_env.sh的

Zhefan-Xu, 2025-04-22:
Your ```ISAACSIM_PATH``` is supposed to be ```/home/lcq/isaac-sim``` instead of ```/home/lcq/isaac-sim/isaac_sim-2023.1.0-hotfix.1```.

charkyli, 2025-04-23:
> Your `ISAACSIM_PATH` is supposed to be `/home/lcq/isaac-sim` instead of `/home/lcq/isaac-sim/isaac_sim-2023.1.0-hotfix.1`.
> 
> Please change the environment variable, delete the conda env by:
> 
> ```
> conda remove --name NavRL --all
> ```
> 
> and redo the installation process.
已经解决了，感谢徐博士！！