Zhefan-Xu/NavRL #52: 训练模型用于机械狗，应该可以吧

Gooddz1, 2025-04-22:
训练模型用于机械狗，应该可以吧？代码修改处应该挺多的吧

Zhefan-Xu, 2025-04-22:
Sure. Please check ROS2 deployment and it is mainly designed for ground robot (e.g. Legged robots).

Gooddz1, 2025-04-23:
在你提供的isaac sim软件中没找到机械狗的结构模型，你们删掉了吗？还是本身就没有的

Gooddz1, 2025-04-23:
好的，谢谢

Gooddz1, 2025-04-23:
算法输出的是机器整体的朝向速度，那么只要设置好环境，用无人机训练好的模型也可以用在机器狗上的吧？

Zhefan-Xu, 2025-04-23:
Sure, we have tested in simulation with a unitree Go2 with the model trained from drones.

Gooddz1, 2025-04-23:
谢谢

Gooddz1, 2025-04-23:
我先加载NavRL-main/quick-demos/ckpts/navrl_checkpoint.pt，再运行python train.py进行训练。但是训练了两个小时得到的checkpoint_2000.pt，用它测试发现机器基本在原地上，或者偏离目标位置。为什么出现这个问题？用原本的NavRL-main/quick-demos/ckpts/navrl_checkpoint.pt测试效果是不错。

Gooddz1, 2025-04-23:
请问，一般训练多少次会得到较好的结果？

Gooddz1, 2025-04-23:
> Sure, we have tested in simulation with a unitree Go2 with the model trained from drones.

应该要去掉无人机的高度才用于机器狗上吧？