Zhefan-Xu/NavRL #53: 激光雷达是否会扫描到动态障碍物？

tangjuju, 2025-04-23:
您好！最近也在学习相关的内容，对您的项目十分感兴趣并且品读了您的IROS2025的论文，和我目前的思路有很多相似的地方，但是想问一下在处理静态和动态障碍物时，动态采用直接的位置输入，静态采用激光雷达输入，则在实际复杂场景中，激光雷达是否会扫到动态障碍物，如何处理？

Zhefan-Xu, 2025-04-23:
First of all, thanks for your interest. Please let me explain the obstacle representation logic:
1. The dynamic obstacle will be detected and tracked with their states.
2. Static environments will be mapped by the occupancy voxels (with the dynamic obstacle region being cleaned)

We will pass the dynamic obstacle information to the network and cast rays to the static occupancy map (as the static obstacle input) to the network.

Hope this explanation can help. Thanks.

tangjuju, 2025-04-23:
I get it！ Thanks！