Zhefan-Xu/NavRL #56: How many training sessions usually yield better results?

Gooddz1, 2025-04-25:
How many times does it usually yield better results to run train.exe directly?

Zhefan-Xu, 2025-04-28:
For 4090, at least ~10-15 hours (depend on the difficulties). It can take longer if you want to train a better results using curriculumn learning.

Gooddz1, 2025-04-28:
I trained a model with LiDAR parameters lidar_hres=1.0 and lidar_vbeams=4.0, and it worked fine when loading the trained model. However, when I changed lidar_vbeams to 1.0 (while keeping lidar_hres=1.0) and retrained the model, I encountered the following error during model loading:
···
return F.linear(input, self.weight, self.bias)RuntimeError: mat1 and mat2 shapes cannot be multiplied (1x2880 and 1440x128)
···

Gooddz1, 2025-04-28:
I trained a model with LiDAR parameters lidar_hres=1.0 and lidar_vbeams=4.0, and it worked fine when loading the trained model. However, when I changed lidar_vbeams to 1.0 (while keeping lidar_hres=1.0) and retrained the model, I encountered the following error during model loading:
return F.linear(input, self.weight, self.bias)RuntimeError: mat1 and mat2 shapes cannot be multiplied (1x2880 and 1440x128)

Gooddz1, 2025-04-29:
Discovering that it is not training, the better the model performance. The model saved in the middle may have better performance than the model saved last