Zhefan-Xu/NavRL #58: Performance Issue

AhJay, 2025-04-28:
Hi, I have a question about the model performance.
As training going on, the training return in Wandb is over 10,000 but grows slowly after 800 steps, but the success rate is still about 50%, which is much lower than the 80% success rate when the number of dynamic obstacle is 60. I wonder if I should still continue training will the success rate go up?

AhJay, 2025-04-28:
In you paper, you mentioned the training takes 10 hours ON RTX 4090. I wonder if the training step can reach 50K in your figure in 10 hours? On my RTX 4090, training step is only about 1K in 1 hour with the default training setting. It seems my performance issue arises from the current small training step.

Zhefan-Xu, 2025-04-28:
This one takes exactly 17h 20m 51s and the paper mentioned approximately 10 hours (if you do not run the evaluation and video updating).