Zhefan-Xu/NavRL #63: Errors in training

wenyuanqin, 2025-05-06:
Hello, I encountered such a problem when training the code. But I don't know how it happened. It seems to be caused by a third-party library.

/NavRL/isaac-training/third_party/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/assets/rigid_object/rigid_object.py", line 204, in write_root_pose_to_sim
    self.root_physx_view.set_transforms(root_poses_xyzw, indices=physx_env_ids)
AttributeError: 'NoneType' object has no attribute 'set_transforms'

wenyuanqin, 2025-05-06:
I used the version isaac-sim:2023.1.0-hotfix.1 provided in README.md and trained it in docker, but after I went online and obtained the assets, I got this error.

Zhefan-Xu, 2025-05-06:
Please check this: https://github.com/Zhefan-Xu/NavRL/issues/7