Zhefan-Xu/NavRL #65: ERROR_INCOMPATIBLE_DRIVER

minglou1984, 2025-05-08:
This is a very good work.

when i run "python training/scripts/train.py" in Verify Installation and Run a Training Example, I got an error as shown:

2025-05-08 12:28:30 [821ms] [Error] [carb.graphics-vulkan.plugin] VkResult: ERROR_INCOMPATIBLE_DRIVER
2025-05-08 12:28:30 [822ms] [Error] [carb.graphics-vulkan.plugin] vkCreateInstance failed. Vulkan 1.1 is not supported, or your driver requires an update.
2025-05-08 12:28:30 [822ms] [Error] [gpu.foundation.plugin] carb::graphics::createInstance failed.
2025-05-08 12:28:30 [1,363ms] [Error] [carb.graphics-vulkan.plugin] VkResult: ERROR_INCOMPATIBLE_DRIVER
2025-05-08 12:28:30 [1,364ms] [Error] [carb.graphics-vulkan.plugin] vkCreateInstance failed. Vulkan 1.1 is not supported, or your driver requires an update.
2025-05-08 12:28:30 [1,364ms] [Error] [gpu.foundation.plugin] carb::graphics::createInstance failed.
2025-05-08 12:28:31 [1,907ms] [Error] [omni.gpu_foundation_factory.plugin] Failed to create GPU foundation devices for compatibilityMode!

It mean INCOMPATIBLE DRIVER, my NVIDIA DRIVER is NVIDIA-SMI 535.129.03, CUDA Version: 12.4

How can i do it?

minglou1984, 2025-05-08:
[image]

I check the System Requirements of Isaac Sim, which are not supported for A100. I use A100-PCIE-40G, this mean i can't run issac sim in this GPU?

Zhefan-Xu, 2025-05-08:
Unfortunately, there is nothing we can do if Isaac Sim do not support A100 and H100... From my experience, RTX 4090 should be enough for training if that is also an option for you.

minglou1984, 2025-05-10:
Thanks for your reply.

My CUDA Toolkit 11.5, Driver 12.4. May be this is INCOMPATIBLE with CUDA driver. I want to try a new CUDA Driver. But I don't know what version of CUDA Driver is ok. Can you provide correctly CUDA version and corresponding driver？

Zhefan-Xu, 2025-05-10:
I tested with 11.8 and 12.2. I believe 11.8+ should all work.