Zhefan-Xu/NavRL #71: Has anyone tried training with variable altitude？

leeexiaoyao, 2025-05-14:
Hi, thanks for the great work!

I noticed that the current training setup seems to assume a fixed altitude during drone flight. But code seems to allow the drones to fly at variable altitude.  Has anyone tried or is there any plan to support training with variable altitude (i.e., allowing the drone to fly at different heights)? I’m particularly interested in safe navigation in 3D environments where altitude changes are necessary.

Any suggestions or references would be greatly appreciated!

Thanks!

Zhefan-Xu, 2025-05-14:
Hi, the training actually allows the robot to move in 3D and the only restriction is a height penalty which prevents the robot flying too high to avoid obstacles.

tanerjn, 2026-11-14:
@Zhefan-Xu Could you point into right direction of the file where height penalty can be adjusted?