Zhefan-Xu/NavRL #72: issue with setup.sh

paridhinaithani, 2025-05-14:
(base) uav@surbhi:~/NavRL/isaac-training$ bash setup.sh 
WARNING: A conda environment already exists at '/home/uav/anaconda3/envs/NavRL'
Remove existing environment (y/[n])? y

Collecting package metadata (current_repodata.json): done
Solving environment: done

==> WARNING: A newer version of conda exists. <==
  current version: 4.8.2
  latest version: 25.3.1

Please update conda by running

    $ conda update -n base -c defaults conda

## Package Plan ##

  environment location: /home/uav/anaconda3/envs/NavRL

  added / updated specs:
    - python=3.10

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    _libgcc_mutex-0.1          |             main           3 KB
    _openmp_mutex-5.1          |            1_gnu          21 KB
    ------------------------------------------------------------
                                           Total:          24 KB

The following NEW packages will be INSTALLED:

  _libgcc_mutex      pkgs/main/linux-64::_libgcc_mutex-0.1-main
  _openmp_mutex      pkgs/main/linux-64::_openmp_mutex-5.1-1_gnu
  bzip2              pkgs/main/linux-64::bzip2-1.0.8-h5eee18b_6
  ca-certificates    pkgs/main/linux-64::ca-certificates-2025.2.25-h06a4308_0
  ld_impl_linux-64   pkgs/main/linux-64::ld_impl_linux-64-2.40-h12ee557_0
  libffi             pkgs/main/linux-64::libffi-3.4.4-h6a678d5_1
  libgcc-ng          pkgs/main/linux-64::libgcc-ng-11.2.0-h1234567_1
  libgomp            pkgs/main/linux-64::libgomp-11.2.0-h1234567_1
  libstdcxx-ng       pkgs/main/linux-64::libstdcxx-ng-11.2.0-h1234567_1
  libuuid            pkgs/main/linux-64::libuuid-1.41.5-h5eee18b_0
  ncurses            pkgs/main/linux-64::ncurses-6.4-h6a678d5_0
  openssl            pkgs/main/linux-64::openssl-3.0.16-h5eee18b_0
  pip                pkgs/main/noarch::pip-25.1-pyhc872135_2
  python             pkgs/main/linux-64::python-3.10.16-he870216_1
  readline           pkgs/main/linux-64::readline-8.2-h5eee18b_0
  setuptools         pkgs/main/linux-64::setuptools-78.1.1-py310h06a4308_0
  sqlite             pkgs/main/linux-64::sqlite-3.45.3-h5eee18b_0
  tk                 pkgs/main/linux-64::tk-8.6.14-h39e8969_0
  tzdata             pkgs/main/noarch::tzdata-2025b-h04d1e81_0
  wheel              pkgs/main/linux-64::wheel-0.45.1-py310h06a4308_0
  xz                 pkgs/main/linux-64::xz-5.6.4-h5eee18b_1
  zlib               pkgs/main/linux-64::zlib-1.2.13-h5eee18b_1

Proceed ([y]/n)? y

Downloading and Extracting Packages
_openmp_mutex-5.1    | 21 KB     | ##################################### | 100% 
_libgcc_mutex-0.1    | 3 KB      | ##################################### | 100% 
Preparing transaction: done
Verifying transaction: done
Executing transaction: done
#
# To activate this environment, use
#
#     $ conda activate NavRL
#
# To deactivate an active environment, use
#
#     $ conda deactivate

Setting up Orbit...
Removing existing symbolic link: _isaac_sim
Running orbit.sh setup...
[INFO] Using conda environment name: NavRL                                     
[INFO] Conda environment named 'NavRL' already exists.
[INFO] Installing extra dependencies (this might take a few minutes)...
(base) uav@surbhi:~/NavRL/isaac-training$

Zhefan-Xu, 2025-05-14:
Hi, can you provide details on what is the issue?

hao551, 2025-07-29:
Same problem