Zhefan-Xu/NavRL #73: Torch version during setup.sh

MaJiazhen8696, 2025-05-16:
great work，very NB orz

I follow the step in readme and installed torch v 2.0.1 ( I dont know which package in setup.sh but there is in setup_deployment.sh for

[image]

Zhefan-Xu, 2025-05-18:
Thanks for posting the issue. 

You can use ```bash setup.sh``` or ```bash setup_deployment.sh``` (if you don't want to train) to set up the virtual environment.

DHU-zyc, 2025-06-26:
I am facing dependency conflicts with the following packages in the NavRL framework:

1. **torch 2.7.1** is incompatible with **torchaudio 2.0.2** and **torchvision 0.15.2**, which require **torch 2.0.1**.
2. **tensordict 0.4.0** requires **torch >= 2.3.0**, but **torchrl 0.8.1** requires **tensordict >= 0.8.3**, which is causing version mismatches.

How can I resolve these conflicts and ensure compatibility with all dependencies for running NavRL demos?