Zhefan-Xu/NavRL #74: How to train on remote server without monitor

MaJiazhen8696, 2025-05-17:
I can train on my pc
When I train on server with headless=true, it still requites a DISPLAY,  maybe vulkan couldn't render correctly

```
[Warning] [omni.isaac.kit.simulation_app] Modules: ['omni_drones', 'omni_drones.utils', 'omni_drones.utils.torchrl.env', 'omni_drones.utils.torchrl.collector', 'omni_drones.utils.torchrl', 'omni_drones.utils.torch', 'omni_drones.controllers.dsl_pid_controller', 'omni_drones.controllers.lee_position_controller', 'omni_drones.controllers', 'omni_drones.utils.torchrl.transforms'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Starting kit application with the following args:  ['/root/Project/NavRL/Env/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/root/Project/NavRL/NavRL/isaac-training/third_party/orbit/_isaac_sim/apps/omni.isaac.sim.python.kit', '--/app/tokens/exe-path=/root/Project/NavRL/NavRL/isaac-training/third_party/orbit/_isaac_sim/kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True', '--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=True', '--/app/fastShutdown=True', '--ext-folder', '/root/Project/NavRL/NavRL/isaac-training/third_party/orbit/_isaac_sim/exts', '--ext-folder', '/root/Project/NavRL/NavRL/isaac-training/third_party/orbit/_isaac_sim/apps', '--/physics/cudaDevice=0', '--portable', '--no-window', '--allow-root']
Passing the following args to the base kit application:  ['headless=True']
[Info] [carb] Logging to file: /root/Project/NavRL/Env/kit/logs/Kit/Isaac-Sim/2023.1/kit_20250517_183626.log
2025-05-17 18:36:26 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[0.122s] [ext: omni.kit.async_engine-0.0.0] startup
[2025-05-17 18:36:27,343][asyncio][DEBUG] - Using selector: EpollSelector
[0.842s] [ext: omni.activity.core-1.0.1] startup
[0.848s] [ext: omni.assets.plugins-0.0.0] startup
[0.851s] [ext: omni.stats-0.0.0] startup
[0.852s] [ext: omni.client-1.0.1] startup
[0.863s] [ext: omni.activity.profiler-1.0.2] startup
[0.866s] [ext: omni.gpu_foundation-0.0.0] startup
[0.881s] [ext: omni.rtx.shadercache.vulkan-1.0.0] startup
[0.883s] [ext: carb.windowing.plugins-1.0.0] startup
2025-05-17 18:36:27 [845ms] [Warning] [carb.windowing-glfw.plugin] GLFW initialization failed.
2025-05-17 18:36:27 [845ms] [Warning] [carb] Failed to startup plugin carb.windowing-glfw.plugin (interfaces: [carb::windowing::IGLContext v1.0],[carb::windowing::IWindowing v1.4]) (impl: carb.windowing-glfw.plugin)
[0.885s] [ext: omni.kit.renderer.init-0.0.0] startup
2025-05-17 18:36:28 [1,870ms] [Warning] [omni.platforminfo.plugin] failed to open the default display.  Can't verify X Server version.
2025-05-17 18:36:28 [2,353ms] [Error] [carb.graphics-vulkan.plugin] VkResult: ERROR_INITIALIZATION_FAILED
2025-05-17 18:36:28 [2,353ms] [Error] [carb.graphics-vulkan.plugin] vkEnumeratePhysicalDevices failed. No physical device is found.
2025-05-17 18:36:28 [2,353ms] [Error] [carb.graphics-vulkan.plugin] No physical device is found.
2025-05-17 18:36:28 [2,354ms] [Error] [gpu.foundation.plugin] carb::graphics::createInstance failed.
2025-05-17 18:36:29 [2,952ms] [Error] [carb.graphics-vulkan.plugin] VkResult: ERROR_INITIALIZATION_FAILED
2025-05-17 18:36:29 [2,953ms] [Error] [carb.graphics-vulkan.plugin] vkEnumeratePhysicalDevices failed. No physical device is found.
2025-05-17 18:36:29 [2,953ms] [Error] [carb.graphics-vulkan.plugin] No physical device is found.
2025-05-17 18:36:29 [2,954ms] [Error] [gpu.foundation.plugin] carb::graphics::createInstance failed.
2025-05-17 18:36:30 [3,729ms] [Error] [omni.gpu_foundation_factory.plugin] Failed to create GPU foundation devices for compatibilityMode!
[3.768s] [ext: omni.kit.pipapi-0.0.0] startup
[3.770s] [ext: omni.kit.pip_archive-0.0.0] startup
[3.771s] [ext: omni.pip.compute-1.2.0] startup
[3.771s] [ext: omni.pip.torch-2_0_1-2.0.2] startup
[3.802s] [ext: omni.pip.cloud-1.0.1] startup
[3.803s] [ext: omni.isaac.core_archive-2.2.1] startup
[3.804s] [ext: omni.isaac.ml_archive-1.1.3] startup
[3.804s] [ext: omni.kit.loop-isaac-1.1.0] startup
[3.805s] [ext: omni.kit.test-0.0.0] startup
[3.807s] [ext: omni.kit.telemetry-0.5.0] startup
2025-05-17 18:36:30 [3,788ms] [Warning] [omni.platforminfo.plugin] failed to load XRAndR.
[2025-05-17 18:36:30,333][omni.kit.telemetry.impl.sentry_extension][INFO] - sentry is disabled for external build
[2025-05-17 18:36:30,333][omni.kit.telemetry.impl.sentry_extension][INFO] - sentry is disabled for external build
[3.832s] [ext: omni.mtlx-0.1.0] startup
[3.833s] [ext: omni.usd.config-1.0.3] startup
[3.840s] [ext: omni.gpucompute.plugins-0.0.0] startup
[3.841s] [ext: omni.usd.libs-1.0.0] startup
[3.992s] [ext: omni.appwindow-1.1.5] startup
2025-05-17 18:36:30 [3,955ms] [Warning] [carb.windowing-glfw.plugin] GLFW initialization failed.
2025-05-17 18:36:30 [3,955ms] [Warning] [carb] Failed to startup plugin carb.windowing-glfw.plugin (interfaces: [carb::windowing::IGLContext v1.0],[carb::windowing::IWindowing v1.4]) (impl: carb.windowing-glfw.plugin)
[3.997s] [ext: omni.kit.renderer.core-0.0.0] startup
2025-05-17 18:36:30 [3,961ms] [Error] [omni.kit.renderer.plugin] GPU Foundation is not initialized!
2025-05-17 18:36:30 [3,967ms] [Warning] [carb.windowing-glfw.plugin] GLFW initialization failed.
2025-05-17 18:36:30 [3,967ms] [Warning] [carb] Failed to startup plugin carb.windowing-glfw.plugin (interfaces: [carb::windowing::IGLContext v1.0],[carb::windowing::IWindowing v1.4]) (impl: carb.windowing-glfw.plugin)
[4.009s] [ext: omni.kit.renderer.capture-0.0.0] startup
[4.012s] [ext: omni.kit.renderer.imgui-0.0.0] startup
2025-05-17 18:36:30 [3,978ms] [Warning] [carb.windowing-glfw.plugin] GLFW initialization failed.
2025-05-17 18:36:30 [3,978ms] [Warning] [carb] Failed to startup plugin carb.windowing-glfw.plugin (interfaces: [carb::windowing::IGLContext v1.0],[carb::windowing::IWindowing v1.4]) (impl: carb.windowing-glfw.plugin)
2025-05-17 18:36:30 [3,979ms] [Error] [omni.kit.imgui_renderer.plugin] _prepareDefaultWindow: invalid device attached to the default window!
[4.022s] [ext: carb.audio-0.1.0] startup
[4.064s] [ext: omni.ui-2.18.1] startup
[4.089s] [ext: omni.uiaudio-1.0.0] startup
[4.091s] [ext: omni.kit.mainwindow-1.0.1] startup
[4.093s] [ext: omni.kit.uiapp-0.0.0] startup
[4.093s] [ext: omni.usd.schema.physics-0.0.0] startup
[4.235s] [ext: omni.usd.schema.isaac-2.0.1] startup
[4.248s] [ext: omni.usd.schema.omnigraph-1.0.0] startup
[... 274 lines cut ...]
[8.648s] [ext: omni.kit.viewport.menubar.camera-105.1.8] startup
[8.656s] [ext: omni.isaac.franka-0.4.0] startup
[8.661s] [ext: omni.kit.viewport.menubar.render-105.1.0] startup
[8.665s] [ext: omni.kit.viewport.menubar.settings-105.0.9] startup
[8.673s] [ext: omni.replicator.composer-1.2.17] startup
[8.692s] [ext: omni.replicator.isaac-1.10.1] startup
[8.734s] [ext: omni.replicator.replicator_yaml-2.0.4] startup
[8.753s] [ext: omni.isaac.cortex-0.3.5] startup
[8.757s] [ext: omni.rtx.settings.core-0.5.12] startup
[8.766s] [ext: omni.kit.viewport.bundle-104.0.1] startup
[8.766s] [ext: semantics.schema.editor-0.3.4] startup
[8.771s] [ext: omni.kit.viewport.menubar.lighting-105.0.7] startup
[8.776s] [ext: omni.isaac.universal_robots-0.3.2] startup
[8.781s] [ext: omni.isaac.cortex.sample_behaviors-1.0.4] startup
[8.786s] [ext: omni.importer.urdf-1.1.3] startup
[8.858s] [ext: omni.kit.window.stats-0.1.4] startup
[8.861s] [ext: omni.isaac.sim.python-2023.1.0] startup
[8.864s] Simulation App Starting
2025-05-17 18:36:35 [9,194ms] [Warning] [omni.hydratexture.plugin] rtx::raytracing::RaycastQueryManager is not available
2025-05-17 18:36:35 [9,233ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] duplicated action as SPACE (On Press) with omni.kit.widget.toolbar.omni.kit.widget.toolbar::toolbar::play!
2025-05-17 18:36:35 [9,233ms] [Warning] [omni.kit.hotkeys.core.registry] [Hotkey] Cannot register Global [omni.kit.widget.toolbar] omni.kit.widget.toolbar::toolbar::play.SPACE, error code: Duplicated action definition
[9.277s] app ready
2025-05-17 18:36:36 [9,721ms] [Warning] [omni.kit.imgui_renderer.plugin] _createExtendCursor: No windowing.
2025-05-17 18:36:36 [9,721ms] [Warning] [omni.kit.imgui_renderer.plugin] _createExtendCursor: No windowing.
```

Zhefan-Xu, 2025-05-18:
You can set the ```headless=True``` when running the training or modify it in the config file.

MaJiazhen8696, 2025-05-18:
Thanks for reply, acutally, the question is Vulkan can't correctly identify the GPU. I configured Vulkan according to 
[https://www.autodl.com/docs/vulkan/](url)
and it can run successfully, but the training speed is a little slow. Now it's running for 8 hours, but wandb only has 10K data. I don't know whether it's normal

Zhefan-Xu, 2025-05-19:
It doesn't look too slow. I am not sure how it compares with 4090 though.