Zhefan-Xu/NavRL #78: demo部分的训练问题

jiexu2358, 2025-05-24:
大佬你好，刚接触强化学习不久，根据你的代码尝试弄了下训练部分，大部分从你的其余部分扒下来和用AI搞的，但是训练很久都不会收敛，想请教一下是否写的有问题，代码如下：
for episode in range(EPISODES):
    # 采样起点终点
    goal = sample_free_goal(obstacles, obstacle_region_min=OBSTACLE_REGION_MIN, obstacle_region_max=OBSTACLE_REGION_MAX)
    robot_pos = sample_free_start(obstacles, goal, obstacle_region_min=OBSTACLE_REGION_MIN, obstacle_region_max=OBSTACLE_REGION_MAX)
    robot_vel = np.array([0.0, 0.0])
    target_dir = goal - robot_pos

    episode_reward = 0

    for step in range(MAX_STEPS):
        # 1. 构造观测
        robot_state = get_robot_state(robot_pos, goal, robot_vel, target_dir, device=device)
        static_obs_input, range_matrix, ray_segments = get_ray_cast(
            robot_pos, obstacles, max_range=MAX_RAY_LENGTH,
            hres_deg=HRES_DEG, vfov_angles_deg=VFOV_ANGLES_DEG,
            start_angle_deg=np.degrees(np.arctan2(target_dir[1], target_dir[0])),
            device=device
        )
        dyn_obs_input = torch.zeros((1, 1, 5, 10), dtype=torch.float, device=device)
        target_dir_tensor = torch.tensor(np.append(target_dir[:2], 0.0), dtype=torch.float, device=device).unsqueeze(0).unsqueeze(0)
        
        done = False
        # 2. 选择动作（带 exploration）
        action, logprob = agent.select_action(robot_state, static_obs_input, dyn_obs_input, target_dir_tensor)
        # 假设action为速度向量
        velocity = action

        # 3. 推进环境
        next_robot_pos = robot_pos + velocity * DT # 更新位置 DT为时间步长 ，等于0.2s
        next_robot_vel = velocity.copy()
        next_target_dir = goal - next_robot_pos

        # 4. 计算奖励（可自行设计：如距离减少为正奖励、碰撞为负等）
        dist_to_goal = np.linalg.norm(goal - next_robot_pos)
        reward = 0.0

        # 4.1 距离奖励
        prev_dist = np.linalg.norm(goal - robot_pos)
        dist_reward = prev_dist - dist_to_goal
        reward += dist_reward * 5.0  # 距离减少的奖励

        # 4.2 碰撞检测和惩罚
        collision, min_dist = check_collision(next_robot_pos, obstacles)
        if collision:
            reward -= 20.0  # 碰撞惩罚
            done = True
        else:
            # 4.3 接近障碍物的惩罚（可选）
            if min_dist < 1.0:  # 当距离障碍物小于1米时
                reward -= (1.0 - min_dist) * 5.0  # 距离越近惩罚越大

        # 4.4 到达目标奖励
        if dist_to_goal < GOAL_REACHED_THRESHOLD:
            reward += 100.0
            done = True

        # 4.5 超出边界惩罚
        if (abs(next_robot_pos[0]) > MAP_HALF_SIZE or 
            abs(next_robot_pos[1]) > MAP_HALF_SIZE):
            reward -= 30.0
            done = True

        # 4.6 步数惩罚（可选）
        reward -= 0.1  # 每步小惩罚，鼓励快速到达目标

        # 5. 存储transition
        agent.store_transition(
            robot_state, static_obs_input, dyn_obs_input, target_dir_tensor,
            action, logprob, reward, done
        )

        episode_reward += reward

        # 6. 状态更新
        robot_pos = next_robot_pos
        robot_vel = next_robot_vel
        target_dir = next_target_dir

        if done:
            break

    # 7. 回合结束后更新策略
    agent.update()  # 按照代码中部分写的agent实现

def update(self):
        # 这里需根据你的 PPO 实现，整理 memory 为 batch，做一次优化
        if len(self.memory) == 0:
            return

        # 整理数据为批量Tensor
        # 先将所有数据转换为numpy数组，然后再一次性转换为tensor
        batch = {}
        for key in self.memory[0]:
            if key != 'done':
                # 先将所有数据转换为numpy数组
                numpy_data = np.array([transition[key] for transition in self.memory])
                # 然后一次性转换为tensor
                batch[key] = torch.tensor(numpy_data, dtype=torch.float32, device=self.device)
        
        # 单独处理done标志
        batch['done'] = torch.tensor([transition['done'] for transition in self.memory], dtype=torch.float32, device=self.device)

        # 假设你的 PPO 支持 update(batch)
        self.policy.update(batch)

        # 清空 memory
        self.memory = []

#policy.update
   def update(self, batch):
        device = self.device

        # 构造 observation tensordict
        obs = TensorDict({
            "agents": TensorDict({
                "observation": TensorDict({
                    "state": batch["robot_state"].squeeze(1),
                    "lidar": batch["static_obs_input"].squeeze(2),
                    "direction": batch["target_dir"],
                    "dynamic_obstacle": batch["dyn_obs_input"].squeeze(2),
                })
            })
        }, device=device)

        actions = batch["action"]
        old_logprobs = batch["logprob"]
        rewards = batch["reward"]
        dones = batch["done"]

        # 1. Forward: 得到 state value
        self.feature_extractor(obs)
        self.critic(obs)
        state_values = obs["state_value"].squeeze(-1)  # [1, B, 1] -> [1, B]
        
        # 确保state_values有正确的维度
        if state_values.dim() == 1:
            state_values = state_values.unsqueeze(0)
        elif state_values.dim() == 3:
            state_values = state_values.squeeze(0)

        # 计算 next_value
        next_state_values = torch.roll(state_values, shifts=-1, dims=0)
        # 修复维度不匹配问题
        next_state_values[0, dones.bool()] = 0.0  # 使用[0, dones.bool()]来匹配[1, B]的形状

        # 2. 计算 GAE Advantage 和 returns
        advantages, returns = self.gae(
            reward=rewards.unsqueeze(0),
            value=state_values,
            terminated=dones.unsqueeze(0),
            next_value=next_state_values,
        )
        
        # 优化张量转换
        advantages = advantages.clone().detach().to(device=device).squeeze(0)
        returns = returns.clone().detach().to(device=device).squeeze(0)

        # 3. 标准化 advantage
        advantages = (advantages - advantages.mean()) / (advantages.std() + 1e-8)

        # 4. 使用整个轨迹段进行训练
        for epoch in range(self.cfg.training_epoch_num):
            # Forward
            out = self(obs)
            new_actions = out["agents", "action"][0]
            action_dist = out["agents", "action_normalized"]
            new_logprobs = out["sample_log_prob"][0]
            state_values = out["state_value"].squeeze(-1).squeeze(0)

            # Policy loss (PPO clipped loss)
            ratio = torch.exp(new_logprobs - old_logprobs)
            surr1 = ratio * advantages
            surr2 = torch.clamp(ratio, 1.0 - self.cfg.actor.clip_ratio, 1.0 + self.cfg.actor.clip_ratio) * advantages
            policy_loss = -torch.min(surr1, surr2).mean()

            # Value loss
            value_loss = self.critic_loss_fn(state_values, returns)
            
            # Entropy loss
            entropy_loss = -action_dist.entropy().mean() if hasattr(action_dist, "entropy") else 0.0

            total_loss = policy_loss + 0.5 * value_loss + self.cfg.entropy_loss_coefficient * entropy_loss

            # 优化器步骤
            self.feature_extractor_optim.zero_grad()
            self.actor_optim.zero_grad()
            self.critic_optim.zero_grad()
            total_loss.backward()
            nn.utils.clip_grad_norm_(self.actor.parameters(), 0.5)
            nn.utils.clip_grad_norm_(self.critic.parameters(), 0.5)
            self.feature_extractor_optim.step()
            self.actor_optim.step()
            self.critic_optim.step()

minglou1984, 2025-07-12:
> 好的，非常感谢，我自己后面也尝试改了，局部坐标系下训练比全局坐标快了一百倍。

你好，局部坐标系下训练，意思是用quick demo还是isaac sim环境呢

jieXu24, 2025-07-19:
> > 好的，非常感谢，我自己后面也尝试改了，局部坐标系下训练比全局坐标快了一百倍。
> 
> 你好，局部坐标系下训练，意思是用quick demo还是isaac sim环境呢

demo，我是做演示用的，后面改为另外的环境了，条件不够，用不了isaac，当时没有改为局部坐标。