Zhefan-Xu/NavRL #82: RuntimeError: CUDA error: no kernel image is available for execution on the device

Mming02, 2025-05-29:
when i run python training/scripts/train.py, there is an error:

[image]

(RuntimeError: CUDA error: no kernel image is available for execution on the device
CUDA kernel errors might be asynchronously reported at some other API call, so the stacktrace below might be incorrect.
For debugging consider passing CUDA_LAUNCH_BLOCKING=1.
Compile with TORCH_USE_CUDA_DSA to enable device-side assertions.
Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.)

my running platform:5070ti  Torch: 2.0.1+cu117 
The workaround I found was to download the PyTorch Nightly version and a newer CUDA version, but the newer version of Torch would have compatibility issues with many dependent libraries. So, can you give me some new ideas for the project to run on a relatively new platform? 
Thank you so much！

Mming02, 2025-05-30:
now , i use torch2.8.0.dev20250529+cu128 which support rtx 5070ti and successfully run the quick demo. But in the training process, as the isaac-sim-orbit strongly rely on torch==2.0.1 
(omni-isaac-orbit 0.15.1 requires torch==2.0.1, but you have torch 2.8.0.dev... which is incompatible.  
omni-isaac-orbit-tasks 0.6.0 requires torch==2.0.1, but you have torch 2.8.0.dev...)  , the training is still failed. So ,I want to know whether there has been success in training with a higher version of torch(>2.0.1),or only version 2.0.1 of torch can works.

Zhefan-Xu, 2025-05-30:
Could you check whether Isaac Sim/orbit supports RTX50 GPU?  I can provide limited help since I currently don't have a 50 GPU on hand

Mming02, 2025-06-04:
surely, I've tested that RTX50 GPUs can run isaac sim/orbit successfully, and my torch+cuda version is 2.8.0.dev20250529+cu128

closed 2025-06-04 by Mming02