Zhefan-Xu/NavRL #87: graphics library issue in the X Window System (X11)

gaohaowe, 2025-06-04:
[image]

When I finished setting up the environment and ran train.py, the terminal reported the error shown above. As far as I know, it is related to the underlying graphics library issue in the X Window System (X11). If you know how to resolve this, I would be truly grateful.

Zhefan-Xu, 2025-06-07:
Thanks for posting the issue. Is this running from a Windows system?

pangruitian, 2025-06-07:
> Thanks for posting the issue. Is this running from a Windows system?

no,Linux ubuntu20.04

MoonGreat, 2025-06-07:
> Thanks for posting the issue. Is this running from a Windows system?

Thanks for your reply. I'm using Ubuntu 20.04 with a nvidia 4090 GPU and the GPU driver version is nvidia-driven-535. Could this be related to the driver version? And what's your driver version?

Zhefan-Xu, 2025-06-07:
I think CUDA driver version might not be the issue since I use various CUDA driver version for example 535, 575

Zhefan-Xu, 2025-06-08:
I've changed the line of creating conda environment from:
```
conda create -n $ENV_NAME python=3.10

```
to 
```
conda create -n $ENV_NAME python=3.10 -c conda-forge
```

It seems conda has changed its defaults channel which adds the following packages that causes errors:
```
+defaults/linux-64::libxcb-1.17.0-h9b100fa_0
+defaults/linux-64::xorg-libx11-1.8.12-h9b100fa_1
+defaults/linux-64::xorg-libxau-1.0.12-h9b100fa_0
+defaults/linux-64::xorg-libxdmcp-1.1.5-h9b100fa_0
+defaults/linux-64::xorg-xorgproto-2024.1-h5eee18b_1
```

The latest commit should address this issue. Please let me know if the issue persists.

MoonGreat, 2025-06-08:
> I've changed the line of creating conda environment from:
>
> ```
> conda create -n $ENV_NAME python=3.10
> ```
>
> to
>
> ```
> conda create -n $ENV_NAME python=3.10 -c conda-forge
> ```
>
> It seems conda has changed [...]

Thanks for your reply appreciately. It's amazing that this operation no longer causes  "python segmentation fault".  However, instead of outputting obstacle information, the isaac sim window keeps dark empty and prompting "Force quit or wait." But after repeated attempts , the normal  scene finally comes out. Thank you very much to you for the solution. 

[image]

[image]

reeceyangzdn, 2025-06-09:
为什么我一开始也是黑屏，但是过一会就自动退出了，试了好几次都是这样。

[image]

[image]

reeceyangzdn, 2025-06-11:
> 我在虚拟环境中将libxcb的版本从1.17降至1.15可以解决这个问题

👍👍👍感谢

deerxiaoxiao, 2025-06-29:
> 为什么我一开始也是黑屏，但是过一会就自动退出了，试了好几次都是这样。
> 
> [image]
> 
> [image]

你好，请问一下你是通过将libxcb的版本从1.17降至1.15解决的吗？我目前也存在这个问题（Isaac sim黑屏加载一会没有出现障碍物，然后自动退出了）。
但我通过conda create -n $ENV_NAME python=3.10 -c conda-forge在setup.sh文件中运行，NavRL环境中已经不存在libxcb了。
而通过conda create -n $ENV_NAME python=3.10会出现核心转储错误，删除libxcb xorg-libx11 xorg-libxau xorg-libxdmcp xorg-xorgproto之后解决核心转储问题，但无论libxcb的版本是1.17还是1.15，依旧存在 Isaac sim黑屏加载一会没有出现障碍物，然后自动退出 的问题。

[image]

[image]

hao551, 2025-08-15:
> > 我在虚拟环境中将libxcb的版本从1.17降至1.15可以解决这个问题
> 
> 🐮阿 🐮阿

conda install libxcb=1.15 -c conda-forge
Traceback (most recent call last):
  File "/home/robot/anaconda3/lib/python3.8/site-packages/conda/exceptions.py", line 1079, in __call__
    return func(*args, **kwargs)
  File "/home/robot/anaconda3/lib/python3.8/site-packages/conda/cli/main.py", line 84, in _main
    exit_code = do_call(args, p)
  File "/home/robot/anaconda3/lib/python3.8/site-packages/conda/cli/conda_argparse.py", line 81, in do_call
    module = import_module(relative_mod, __name__.rsplit('.', 1)[0])
  File "/home/robot/anaconda3/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/robot/anaconda3/lib/python3.8/site-packages/conda/cli/main_install.py", line 8, in <module>
    from .install import install
  File "/home/robot/anaconda3/lib/python3.8/site-packages/conda/cli/install.py", line 19, in <module>
    from ..core.index import calculate_channel_urls, get_index
  File "/home/robot/anaconda3/lib/python3.8/site-packages/conda/core/index.py", line 13, in <module>
    from .package_cache_data import PackageCacheData
  File "/home/robot/anaconda3/lib/python3.8/site-packages/conda/core/package_cache_data.py", line 14, in <module>
    from .path_actions import CacheUrlAction, ExtractPackageAction
  File "/home/robot/anaconda3/lib/python3.8/site-packages/conda/core/path_actions.py", line 30, in <module>
    from ..gateways.connection.download import download
  File "/home/robot/anaconda3/lib/python3.8/site-packages/conda/gateways/connection/download.py", line 14, in <module>
    from .session import CondaSession
  File "/home/robot/anaconda3/lib/python3.8/site-packages/conda/gateways/connection/session.py", line 13, in <module>
    from .adapters.s3 import S3Adapter
  File "/home/robot/anaconda3/lib/python3.8/site-packages/conda/gateways/connection/adapters/s3.py", line 12, in <module>
    import boto3
  File "/home/robot/isaac-copy/exts/omni.pip.cloud/pip_prebundle/boto3/__init__.py", line 17, in <module>
    from boto3.session import Session
  File "/home/robot/isaac-copy/exts/omni.pip.cloud/pip_prebundle/boto3/session.py", line 17, in <module>
    import botocore.session
  File "/home/robot/isaac-copy/kit/exts/omni.kit.pip_archive/pip_prebundle/botocore/session.py", line 30, in <module>
    import botocore.credentials
  File "/home/robot/isaac-copy/kit/exts/omni.kit.pip_archive/pip_prebundle/botocore/credentials.py", line 34, in <module>
    from botocore.config import Config
  File "/home/robot/isaac-copy/kit/exts/omni.kit.pip_archive/pip_prebundle/botocore/config.py", line 16, in <module>
    from botocore.endpoint import DEFAULT_TIMEOUT, MAX_POOL_CONNECTIONS
  File "/home/robot/isaac-copy/kit/exts/omni.kit.pip_archive/pip_prebundle/botocore/endpoint.py", line 22, in <module>
    from botocore.awsrequest import create_request_object
  File "/home/robot/isaac-copy/kit/exts/omni.kit.pip_archive/pip_prebundle/botocore/awsrequest.py", line 26, in <module>
    import botocore.utils
  File "/home/robot/isaac-copy/kit/exts/omni.kit.pip_archive/pip_prebundle/botocore/utils.py", line 33, in <module>
    import botocore.httpsession
  File "/home/robot/isaac-copy/kit/exts/omni.kit.pip_archive/pip_prebundle/botocore/httpsession.py", line 17, in <module>
    from urllib3.contrib.pyopenssl import orig_util_SSLContext as SSLContext
  File "/home/robot/isaac-copy/kit/exts/omni.kit.pip_archive/pip_prebundle/urllib3/contrib/pyopenssl.py", line 50, in <module>
    import OpenSSL.crypto
  File "/home/robot/anaconda3/lib/python3.8/site-packages/OpenSSL/__init__.py", line 8, in <module>
    from OpenSSL import crypto, SSL
  File "/home/robot/anaconda3/lib/python3.8/site-packages/OpenSSL/crypto.py", line 1556, in <module>
    class X509StoreFlags(object):
  File "/home/robot/anaconda3/lib/python3.8/site-packages/OpenSSL/crypto.py", line 1577, in X509StoreFlags
    CB_ISSUER_CHECK = _lib.X509_V_FLAG_CB_ISSUER_CHECK
AttributeError: module 'lib' has no attribute 'X509_V_FLAG_CB_ISSUER_CHECK'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/robot/anaconda3/bin/conda", line 13, in <module>
    sys.exit(main())
  File "/home/robot/anaconda3/lib/python3.8/site-packages/conda/cli/main.py", line 152, in main
[... 28 lines cut ...]
    import botocore.session
  File "/home/robot/isaac-copy/kit/exts/omni.kit.pip_archive/pip_prebundle/botocore/session.py", line 30, in <module>
    import botocore.credentials
  File "/home/robot/isaac-copy/kit/exts/omni.kit.pip_archive/pip_prebundle/botocore/credentials.py", line 34, in <module>
    from botocore.config import Config
  File "/home/robot/isaac-copy/kit/exts/omni.kit.pip_archive/pip_prebundle/botocore/config.py", line 16, in <module>
    from botocore.endpoint import DEFAULT_TIMEOUT, MAX_POOL_CONNECTIONS
  File "/home/robot/isaac-copy/kit/exts/omni.kit.pip_archive/pip_prebundle/botocore/endpoint.py", line 22, in <module>
[... cut ...]
  File "/home/robot/anaconda3/lib/python3.8/site-packages/OpenSSL/__init__.py", line 8, in <module>
    from OpenSSL import crypto, SSL
  File "/home/robot/anaconda3/lib/python3.8/site-packages/OpenSSL/crypto.py", line 1556, in <module>
    class X509StoreFlags(object):
  File "/home/robot/anaconda3/lib/python3.8/site-packages/OpenSSL/crypto.py", line 1577, in X509StoreFlags
    CB_ISSUER_CHECK = _lib.X509_V_FLAG_CB_ISSUER_CHECK
AttributeError: module 'lib' has no attribute 'X509_V_FLAG_CB_ISSUER_CHECK'
您好，想问下为什么我想降级会报这个错误