Zhefan-Xu/NavRL #89: 在运行train.py时，出现问题“段错误 (核心已转储)”

leeexiaoyao, 2025-06-12:
[image]
运行python training/scripts/train.py时出现的问题

zdytim, 2025-06-19:
参考X11那个issue，是环境更新的问题

zdytim, 2025-06-19:
> > 参考X11那个issue，是环境更新的问题
> 
> 我通过将libxcb的版本从1.17降至1.15解决了此问题，但随之而来的是另一个问题
> 
> [image]
> 
> [image]

https://github.com/Zhefan-Xu/NavRL/issues/87#issuecomment-2953294556 
参考这个，把那几个包删了

leeexiaoyao, 2025-06-21:
> > > > > 参考X11那个issue，是环境更新的问题
> > > > 
> > > > 
> > > > 我通过将libxcb的版本从1.17降至1.15解决了此问题，但随之而来的是另一个问题
> > > > [image]
> > > > [image]
> > > 
> > > 
> > > [#87 (comment)](https://github.com/Zhefan-Xu/NavRL/issues/87#issuecomment-2953294556) 参考这个，把那几个包删了
> > 
> > 
> > 具体来说我是修改了这部分解决了第一步
> > conda create -n $ENV_NAME python=3.10
> > to
> > conda create -n $ENV_NAME python=3.10 -c conda-forge
> > 然后应该是没有这几个包libxcb xorg-libx11 xorg-libxau xorg-libxdmcp xorg-xorgproto --force 然后我又尝试了有无libxcb==1.15两种情况，但都不行，还是一样的问题
> > [image]
> 
> 多等一会试试，你试试后面的训练可不可以，不要上来就去点isaac sim。这只是报警不应该会出现不能运行的问题

我在运行train.py时通过在另一个终端执行
pkill -f isaac-sim
解决了此问题，成功出现了动态障碍物地图画面
谢谢您的帮助，非常感谢！

leeexiaoyao, 2025-06-22:
> this issue how to solve it
> 
> [image]

you can modify the file NavRL/isaac-training/third_party/orbit/source/extensions/omni.isaac.orbit_tasks/omni/isaac/orbit_tasks/locomotion/velocity/velocity_env_cfg.py

leeexiaoyao, 2025-06-22:
> this issue how to solve it
> 
> [image]

you can modify the file NavRL/isaac-training/third_party/orbit/source/extensions/omni.isaac.orbit_tasks/omni/isaac/orbit_tasks/locomotion/velocity/velocity_env_cfg.py    replace the part:visual_material=sim_utils.MdlFileCfg(  
    mdl_path="{NVIDIA_NUCLEUS_DIR}/Materials/Base/Architecture/Shingles_01.mdl",  
    project_uvw=True,  
),  
  
to
visual_material=sim_utils.PreviewSurfaceCfg(  
    diffuse_color=(0.6, 0.5, 0.4),  # 瓦片/地面颜色  
    roughness=0.8,  
    metallic=0.0,  
),

glt621, 2025-06-24:
> > 这个问题该如何解决
> > [image]
> 
> 您可以修改文件 NavRL/isaac-training/third_party/orbit/source/extensions/omni.isaac.orbit_tasks/omni/isaac/orbit_tasks/locomotion/velocity/velocity_env_cfg.py 替换部分：visual_material=sim_utils.MdlFileCfg( mdl_path="{NVIDIA_NUCLEUS_DIR}/Materials/Base/Architecture/Shingles_01.mdl", project_uvw=True, ),
> 
> 到 Visual_material=sim_utils.PreviewSurfaceCfg( diffusive_color=(0.6, 0.5, 0.4), # 瓦片/ Ground Colorroughness =0.8, Metallic=0.0, )，我 通过这一步解决了这个问题。

我找到解决方法了，应该是服务器代理出现问题了，具体是什么原因不清楚。当我把代理关掉，在终端加入代理端口号就正常运行了

hao551, 2025-08-15:
> 参考X11那个issue，是环境更新的问题

conda remove libxcb xorg-libx11 xorg-libxau xorg-libxdmcp xorg-xorgproto --force
Traceback (most recent call last):
  File "/home/robot/anaconda3/lib/python3.8/site-packages/conda/exceptions.py", line 1079, in __call__
    return func(*args, **kwargs)
  File "/home/robot/anaconda3/lib/python3.8/site-packages/conda/cli/main.py", line 84, in _main
    exit_code = do_call(args, p)
  File "/home/robot/anaconda3/lib/python3.8/site-packages/conda/cli/conda_argparse.py", line 81, in do_call
    module = import_module(relative_mod, __name__.rsplit('.', 1)[0])
  File "/home/robot/anaconda3/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/robot/anaconda3/lib/python3.8/site-packages/conda/cli/main_remove.py", line 11, in <module>
    from .install import handle_txn
  File "/home/robot/anaconda3/lib/python3.8/site-packages/conda/cli/install.py", line 19, in <module>
    from ..core.index import calculate_channel_urls, get_index
  File "/home/robot/anaconda3/lib/python3.8/site-packages/conda/core/index.py", line 13, in <module>
    from .package_cache_data import PackageCacheData
  File "/home/robot/anaconda3/lib/python3.8/site-packages/conda/core/package_cache_data.py", line 14, in <module>
    from .path_actions import CacheUrlAction, ExtractPackageAction
  File "/home/robot/anaconda3/lib/python3.8/site-packages/conda/core/path_actions.py", line 30, in <module>
    from ..gateways.connection.download import download
  File "/home/robot/anaconda3/lib/python3.8/site-packages/conda/gateways/connection/download.py", line 14, in <module>
    from .session import CondaSession
  File "/home/robot/anaconda3/lib/python3.8/site-packages/conda/gateways/connection/session.py", line 13, in <module>
    from .adapters.s3 import S3Adapter
  File "/home/robot/anaconda3/lib/python3.8/site-packages/conda/gateways/connection/adapters/s3.py", line 12, in <module>
    import boto3
  File "/home/robot/isaac-copy/exts/omni.pip.cloud/pip_prebundle/boto3/__init__.py", line 17, in <module>
    from boto3.session import Session
  File "/home/robot/isaac-copy/exts/omni.pip.cloud/pip_prebundle/boto3/session.py", line 17, in <module>
    import botocore.session
  File "/home/robot/isaac-copy/kit/exts/omni.kit.pip_archive/pip_prebundle/botocore/session.py", line 30, in <module>
    import botocore.credentials
  File "/home/robot/isaac-copy/kit/exts/omni.kit.pip_archive/pip_prebundle/botocore/credentials.py", line 34, in <module>
    from botocore.config import Config
  File "/home/robot/isaac-copy/kit/exts/omni.kit.pip_archive/pip_prebundle/botocore/config.py", line 16, in <module>
    from botocore.endpoint import DEFAULT_TIMEOUT, MAX_POOL_CONNECTIONS
  File "/home/robot/isaac-copy/kit/exts/omni.kit.pip_archive/pip_prebundle/botocore/endpoint.py", line 22, in <module>
    from botocore.awsrequest import create_request_object
  File "/home/robot/isaac-copy/kit/exts/omni.kit.pip_archive/pip_prebundle/botocore/awsrequest.py", line 26, in <module>
    import botocore.utils
  File "/home/robot/isaac-copy/kit/exts/omni.kit.pip_archive/pip_prebundle/botocore/utils.py", line 33, in <module>
    import botocore.httpsession
  File "/home/robot/isaac-copy/kit/exts/omni.kit.pip_archive/pip_prebundle/botocore/httpsession.py", line 17, in <module>
    from urllib3.contrib.pyopenssl import orig_util_SSLContext as SSLContext
  File "/home/robot/isaac-copy/kit/exts/omni.kit.pip_archive/pip_prebundle/urllib3/contrib/pyopenssl.py", line 50, in <module>
    import OpenSSL.crypto
  File "/home/robot/anaconda3/lib/python3.8/site-packages/OpenSSL/__init__.py", line 8, in <module>
    from OpenSSL import crypto, SSL
  File "/home/robot/anaconda3/lib/python3.8/site-packages/OpenSSL/crypto.py", line 1556, in <module>
    class X509StoreFlags(object):
  File "/home/robot/anaconda3/lib/python3.8/site-packages/OpenSSL/crypto.py", line 1577, in X509StoreFlags
    CB_ISSUER_CHECK = _lib.X509_V_FLAG_CB_ISSUER_CHECK
AttributeError: module 'lib' has no attribute 'X509_V_FLAG_CB_ISSUER_CHECK'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/robot/anaconda3/bin/conda", line 13, in <module>
    sys.exit(main())
  File "/home/robot/anaconda3/lib/python3.8/site-packages/conda/cli/main.py", line 152, in main
    return conda_exception_handler(_main, *args, **kwargs)
  File "/home/robot/anaconda3/lib/python3.8/site-packages/conda/exceptions.py", line 1371, in conda_exception_handler
[... 26 lines cut ...]
    import botocore.session
  File "/home/robot/isaac-copy/kit/exts/omni.kit.pip_archive/pip_prebundle/botocore/session.py", line 30, in <module>
    import botocore.credentials
  File "/home/robot/isaac-copy/kit/exts/omni.kit.pip_archive/pip_prebundle/botocore/credentials.py", line 34, in <module>
    from botocore.config import Config
  File "/home/robot/isaac-copy/kit/exts/omni.kit.pip_archive/pip_prebundle/botocore/config.py", line 16, in <module>
[... cut ...]
  File "/home/robot/anaconda3/lib/python3.8/site-packages/OpenSSL/__init__.py", line 8, in <module>
    from OpenSSL import crypto, SSL
  File "/home/robot/anaconda3/lib/python3.8/site-packages/OpenSSL/crypto.py", line 1556, in <module>
    class X509StoreFlags(object):
  File "/home/robot/anaconda3/lib/python3.8/site-packages/OpenSSL/crypto.py", line 1577, in X509StoreFlags
    CB_ISSUER_CHECK = _lib.X509_V_FLAG_CB_ISSUER_CHECK
AttributeError: module 'lib' has no attribute 'X509_V_FLAG_CB_ISSUER_CHECK'
想请问一下为什么会出现这个情况，删不掉这几个包

hao551, 2025-08-15:
> 多等一会试试，你试试后面的训练可不可以，不要上来就去点isaac sim。这只是报警不应该会出现不能运
[0.615s] [ext: omni.kit.pipapi-0.0.0] startup
[0.617s] [ext: omni.kit.pip_archive-0.0.0] startup
[0.617s] [ext: omni.pip.compute-1.2.0] startup
[0.617s] [ext: omni.pip.torch-2_0_1-2.0.2] startup
[0.645s] [ext: omni.pip.cloud-1.0.1] startup
[0.651s] [ext: omni.isaac.core_archive-2.2.1] startup
[0.652s] [ext: omni.isaac.ml_archive-1.1.3] startup
[0.652s] [ext: omni.kit.loop-isaac-1.1.0] startup
[0.652s] [ext: omni.kit.test-0.0.0] startup
[0.653s] [ext: omni.kit.telemetry-0.5.0] startup
[xcb] Unknown sequence number while appending request
[xcb] You called XInitThreads, this is not your fault
[xcb] Aborting, sorry about that.
[xcb] Extra reply data still left in queue
[xcb] This is most likely caused by a broken X extension library
[xcb] Aborting, sorry about that.
Fatal Python error: Segmentation fault

Thread 0x00007f22b1727180 (most recent call first):
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 1176 in create_module
  File "<frozen importlib._bootstrap>", line 571 in module_from_spec
  File "<frozen importlib._bootstrap>", line 674 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1006 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1027 in _find_and_load
  File "/home/robot/isaac-copy/kit/exts/omni.kit.telemetry/omni/kit/telemetry/__init__.py", line 7 in <module>
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 883 in exec_module
  File "<frozen importlib._bootstrap>", line 688 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1006 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1027 in _find_and_load
  File "<frozen importlib._bootstrap>", line 1050 in _gcd_import
  File "/home/robot/anaconda3/envs/NavRL/lib/python3.10/importlib/__init__.py", line 126 in import_module
  File "/home/robot/isaac-copy/kit/kernel/py/omni/ext/_impl/custom_importer.py", line 76 in import_module
  File "/home/robot/isaac-copy/kit/kernel/py/omni/ext/_impl/_internal.py", line 94 in _import_public
  File "/home/robot/isaac-copy/kit/kernel/py/omni/ext/_impl/_internal.py", line 187 in _custom_importer
  File "/home/robot/isaac-copy/kit/kernel/py/omni/ext/_impl/_internal.py", line 211 in startup
  File "/home/robot/isaac-copy/kit/kernel/py/omni/ext/_impl/_internal.py", line 328 in startup_extension
  File "PythonExtension.cpp::startup()", line 2 in <module>
  File "/home/robot/isaac-copy/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py", line 303 in _start_app
  File "/home/robot/isaac-copy/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py", line 192 in __init__
  File "/home/robot/NavRL/isaac-training/training/scripts/train.py", line 24 in main
  File "/home/robot/anaconda3/envs/NavRL/lib/python3.10/site-packages/hydra/core/utils.py", line 186 in run_job
  File "/home/robot/anaconda3/envs/NavRL/lib/python3.10/site-packages/hydra/_internal/hydra.py", line 119 in run
  File "/home/robot/anaconda3/envs/NavRL/lib/python3.10/site-packages/hydra/_internal/utils.py", line 458 in <lambda>
  File "/home/robot/anaconda3/envs/NavRL/lib/python3.10/site-packages/hydra/_internal/utils.py", line 220 in run_and_report
  File "/home/robot/anaconda3/envs/NavRL/lib/python3.10/site-packages/hydra/_internal/utils.py", line 457 in _run_app
  File "/home/robot/anaconda3/envs/NavRL/lib/python3.10/site-packages/hydra/_internal/utils.py", line 394 in _run_hydra
  File "/home/robot/anaconda3/envs/NavRL/lib/python3.10/site-packages/hydra/main.py", line 94 in decorated_main
  File "/home/robot/NavRL/isaac-training/training/scripts/train.py", line 138 in <module>

Extension modules: yaml._yaml, numpy.core._multiarray_umath, numpy.core._multiarray_tests, numpy.linalg._umath_linalg, numpy.fft._pocketfft_internal, numpy.random._common, numpy.random.bit_generator, numpy.random._bounded_integers, numpy.random._mt19937, numpy.random.mtrand, numpy.random._philox, numpy.random._pcg64, numpy.random._sfc64, numpy.random._generator, h5py._errors, h5py.defs, h5py._objects, h5py.h5, h5py.utils, h5py.h5t, h5py.h5s, h5py.h5ac, h5py.h5p, h5py.h5r, h5py._proxy, h5py._conv, h5py.h5z, h5py.h5a, h5py.h5d, h5py.h5ds, h5py.h5g, h5py.h5i, h5py.h5o, h5py.h5f, h5py.h5fd, h5py.h5pl, h5py.h5l, h5py._selector, google.protobuf.pyext._message, psutil._psutil_linux, psutil._psutil_posix, torch._C, torch._C._fft, torch._C._linalg, torch._C._nested, torch._C._nn, torch._C._sparse, torch._C._special (total: 48)
段错误 (核心已转储)
我在运行train.py时出现了这个