Zhefan-Xu/NavRL #92: 创建虚拟环境过程中遇到的torch版本问题

LiuLu-eng, 2025-06-21:
徐博士你好，最近在尝试复现您这个开源工作时遇到一些问题。当我尝试不训练代理，直接部署虚拟环境时，在运行“cd NavRL/isaac-training        bash setup_deployment.sh”之后，前面的包都正常安装，但是在安装TensorDict和TorchRL时出现如下报错：

[image]
[image]
[image]

请问这个问题应该如何解决？谢谢。

LiuLu-eng, 2025-06-25:
> 同问，解决了吗

我把third_party/tensordict/setup.py 中第70行的【return "torch>=2.1.0"】改为了【return "torch>=2.0.1"】可以成功安装tensordict。 同样的把third_party/rl/setup.py中第76行的【return "torch>=2.1.0"】改为了【return "torch>=2.0.1"】可以成功安装torchrl。

但是配置好虚拟环境后第三句指令：rosrun navigation_runner navigation_node.py运行报错：
(NavRL) zb123@123:~/Navrl_ws$ rosrun navigation_runner navigation_node.py 
/home/zb123/anaconda3/envs/NavRL/lib/python3.10/site-packages/torch/nn/modules/lazy.py:180: UserWarning: Lazy modules are a new feature under heavy development so changes to the API or functionality can happen at any moment.
  warnings.warn('Lazy modules are a new feature under heavy development '
Error executing job with overrides: []
RuntimeError: TensorDictModule failed with operation
    Sequential(
      (0): Conv2d(1, 4, kernel_size=(5, 3), stride=(1, 1), padding=(2, 1))
      (1): ELU(alpha=1.0)
      (2): LazyConv2d(0, 16, kernel_size=(5, 3), stride=(2, 1), padding=(2, 1))
      (3): ELU(alpha=1.0)
      (4): LazyConv2d(0, 16, kernel_size=(5, 3), stride=(2, 2), padding=(2, 1))
      (5): ELU(alpha=1.0)
      (6): Rearrange('n c w h -> n (c w h)')
      (7): LazyLinear(in_features=0, out_features=128, bias=True)
      (8): LayerNorm((128,), eps=1e-05, elementwise_affine=True)
    )
    in_keys=[('agents', 'observation', 'lidar')]
    out_keys=['_cnn_feature'].

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/zb123/Navrl_ws/src/ros1/navigation_runner/scripts/navigation_node.py", line 18, in <module>
    main()
  File "/home/zb123/anaconda3/envs/NavRL/lib/python3.10/site-packages/hydra/main.py", line 94, in decorated_main
    _run_hydra(
  File "/home/zb123/anaconda3/envs/NavRL/lib/python3.10/site-packages/hydra/_internal/utils.py", line 394, in _run_hydra
    _run_app(
  File "/home/zb123/anaconda3/envs/NavRL/lib/python3.10/site-packages/hydra/_internal/utils.py", line 457, in _run_app
    run_and_report(
  File "/home/zb123/anaconda3/envs/NavRL/lib/python3.10/site-packages/hydra/_internal/utils.py", line 223, in run_and_report
    raise ex
  File "/home/zb123/anaconda3/envs/NavRL/lib/python3.10/site-packages/hydra/_internal/utils.py", line 220, in run_and_report
    return func()
  File "/home/zb123/anaconda3/envs/NavRL/lib/python3.10/site-packages/hydra/_internal/utils.py", line 458, in <lambda>
    lambda: hydra.run(
  File "/home/zb123/anaconda3/envs/NavRL/lib/python3.10/site-packages/hydra/_internal/hydra.py", line 132, in run
    _ = ret.return_value
  File "/home/zb123/anaconda3/envs/NavRL/lib/python3.10/site-packages/hydra/core/utils.py", line 260, in return_value
    raise self._return_value
  File "/home/zb123/anaconda3/envs/NavRL/lib/python3.10/site-packages/hydra/core/utils.py", line 186, in run_job
    ret.return_value = task_function(task_cfg)
  File "/home/zb123/Navrl_ws/src/ros1/navigation_runner/scripts/navigation_node.py", line 12, in main
    nav = Navigation(cfg)
  File "/home/zb123/Navrl_ws/src/ros1/navigation_runner/scripts/navigation.py", line 75, in __init__
    self.policy = self.init_model()
  File "/home/zb123/Navrl_ws/src/ros1/navigation_runner/scripts/navigation.py", line 107, in init_model
    policy = PPO(self.cfg.algo, observation_spec, action_spec, self.cfg.device)
  File "/home/zb123/Navrl_ws/src/ros1/navigation_runner/scripts/ppo.py", line 72, in __init__
    self.__call__(dummy_input)
  File "/home/zb123/Navrl_ws/src/ros1/navigation_runner/scripts/ppo.py", line 83, in __call__
    self.feature_extractor(tensordict)
  File "/home/zb123/anaconda3/envs/NavRL/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1501, in _call_impl
    return forward_call(*args, **kwargs)
  File "/home/zb123/NavRL/isaac-training/third_party/tensordict/tensordict/nn/common.py", line 289, in wrapper
    return func(_self, tensordict, *args, **kwargs)
  File "/home/zb123/NavRL/isaac-training/third_party/tensordict/tensordict/_contextlib.py", line 126, in decorate_context
    return func(*args, **kwargs)
  File "/home/zb123/NavRL/isaac-training/third_party/tensordict/tensordict/nn/utils.py", line 261, in wrapper
    return func(_self, tensordict, *args, **kwargs)
  File "/home/zb123/NavRL/isaac-training/third_party/tensordict/tensordict/nn/sequence.py", line 428, in forward
    tensordict = self._run_module(module, tensordict, **kwargs)
  File "/home/zb123/NavRL/isaac-training/third_party/tensordict/tensordict/nn/sequence.py", line 409, in _run_module
    tensordict = module(tensordict, **kwargs)
  File "/home/zb123/anaconda3/envs/NavRL/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1501, in _call_impl
[... 1 lines cut ...]
  File "/home/zb123/NavRL/isaac-training/third_party/tensordict/tensordict/nn/common.py", line 289, in wrapper
    return func(_self, tensordict, *args, **kwargs)
  File "/home/zb123/NavRL/isaac-training/third_party/tensordict/tensordict/_contextlib.py", line 126, in decorate_context
    return func(*args, **kwargs)
  File "/home/zb123/NavRL/isaac-training/third_party/tensordict/tensordict/nn/utils.py", line 261, in wrapper
    return func(_self, tensordict, *args, **kwargs)
  File "/home/zb123/NavRL/isaac-training/third_party/tensordict/tensordict/nn/common.py", line 1224, in forward
    raise err from RuntimeError(
  File "/home/zb123/NavRL/isaac-training/third_party/tensordict/tensordict/nn/common.py", line 1198, in forward
    raise err
[... cut ...]
  File "/home/zb123/anaconda3/envs/NavRL/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1538, in _call_impl
    result = forward_call(*args, **kwargs)
  File "/home/zb123/anaconda3/envs/NavRL/lib/python3.10/site-packages/torch/nn/modules/conv.py", line 463, in forward
    return self._conv_forward(input, self.weight, self.bias)
  File "/home/zb123/anaconda3/envs/NavRL/lib/python3.10/site-packages/torch/nn/modules/conv.py", line 459, in _conv_forward
    return F.conv2d(input, weight, bias, self.stride,
RuntimeError: GET was unable to find an engine to execute this computation

huangxiaobing12, 2025-06-26:
这样修改确实成功了，但是后续由于我的cuda版本是12.8和isaac版本有冲突，还是导致这个.py文件运行失败

LiuLu-eng, 2025-06-26:
> 这样修改确实成功了，但是后续由于我的cuda版本是12.8和isaac版本有冲突，还是导致这个.py文件运行失败

报的错和我的一样么？我的cuda版本为12.2

LiuLu-eng, 2025-06-26:
> 不太一样，我在修改我的cuda版本再试试，我的isaacsim界面能启动但是没有障碍物

如果你修改cuda版本之后成功了，请告诉我你用的是什么版本的，谢谢。