Zhefan-Xu/NavRL #93: PhysX errors

aiyuke111, 2025-06-24:
I'm encountering multiple PhysX errors during training in Isaac Sim with a Hummingbird drone setup. The logs show errors such as: "Transformation change on non-root links is not supported. Link: /World/envs/env_X/Hummingbird_0/rotor_Y" for various environments (e.g., env_28, env_19, env_20, env_22, env_23, env_24, env_25, env_1, env_27, env_29, env_5, env_8) and rotors (e.g., rotor_1, rotor_2, rotor_3). Additionally, there are errors stating: "PxArticulationLink::setGlobalPose(): it is illegal to call this method if PxSceneFlag::eENABLE_DIRECT_GPU_API is enabled!" from NpArticulationLink.cpp, LINE 343, occurring repeatedly at timestamps like 2025-06-24 07:20:33 [156,085ms] to [156,088ms]. These errors appear during environment resets, likely due to calls to set_world_poses in env.py. The simulation continues but is slow in headless=False mode. Can you suggest a fix, such as disabling eENABLE_DIRECT_GPU_API in the simulation configuration or modifying the pose-setting logic to avoid transformations on non-root links like the drone's rotors?  
2025-06-24 07:20:33 [156,085ms] [Error] [omni.physx.plugin] Transformation change on non-root links is not supported. Link: /World/envs/env_27/Hummingbird_0/rotor_2
2025-06-24 07:20:33 [156,085ms] [Error] [omni.physx.plugin] PhysX error: PxArticulationLink::setGlobalPose(): it is illegal to call this method if PxSceneFlag::eENABLE_DIRECT_GPU_API is enabled!, FILE /buildAgent/work/eb2f45c4acc808a0/physx/source/physx/src/NpArticulationLink.cpp, LINE 343
2025-06-24 07:20:34 [156,831ms] [Warning] [omni.kit.notification_manager.manager] PhysX error: PxArticulationLink::setGlobalPose(): it is illegal to call this method if PxSceneFlag::eENABLE_DIRECT_GPU_API is enabled!

hao509913-arch, 2025-11-13:
Has it been solved?

hao509913-arch, 2025-11-13:
i have the same problem