Zhefan-Xu/NavRL #97: 在执行python training/scripts/train.py出错

linlei66, 2025-06-28:
Error executing job with overrides: []
Traceback (most recent call last):
  File "/home/lin/NavRL/isaac-training/training/scripts/train.py", line 49, in main
    env = NavigationEnv(cfg)
  File "/home/lin/NavRL/isaac-training/third_party/rl/torchrl/envs/common.py", line 159, in __call__
    instance: EnvBase = super().__call__(*args, **kwargs)
  File "/home/lin/NavRL/isaac-training/training/scripts/env.py", line 39, in __init__
    super().__init__(cfg, cfg.headless)
  File "/home/lin/NavRL/isaac-training/third_party/OmniDrones/omni_drones/envs/isaac_env.py", line 134, in __init__
    global_prim_paths = self._design_scene() # in the child class, init robot (1), light, terrain (with number of env)
  File "/home/lin/NavRL/isaac-training/training/scripts/env.py", line 100, in _design_scene
    cfg_ground.func("/World/defaultGroundPlane", cfg_ground, translation=(0, 0, 0.01))
  File "/home/lin/NavRL/isaac-training/third_party/orbit/source/extensions/omni.isaac.orbit/omni/isaac/orbit/sim/spawners/from_files/from_files.py", line 145, in spawn_ground_plane
    collision_prim_path = prim_utils.get_prim_path(
  File "/home/lin/isaacsim/exts/omni.isaac.core/omni/isaac/core/utils/prims.py", line 349, in get_prim_path
    return prim.GetPath()
AttributeError: 'NoneType' object has no attribute 'GetPath'

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.
wandb: Waiting for W&B process to finish... (failed 1).
wandb:                                                                                
wandb: You can sync this run to the cloud by running:
wandb: wandb sync /home/lin/NavRL/isaac-training/wandb/offline-run-20250628_101136-d4i9dj88
wandb: Find logs at: ./wandb/offline-run-20250628_101136-d4i9dj88/logs
然后弹出

XXLiu-HNU, 2025-07-15:
相同的问题，请问您解决了吗

XXLiu-HNU, 2025-07-15:
这个是完整的输出：
```
(NavRL) xingxun@R9000:~/experment/NavRL/isaac-training$ python training/scripts/train.py
[Warning] [omni.isaac.kit.simulation_app] Modules: ['omni_drones', 'omni_drones.utils', 'omni_drones.utils.torchrl.env', 'omni_drones.utils.torchrl.collector', 'omni_drones.utils.torchrl', 'omni_drones.utils.torch', 'omni_drones.controllers.dsl_pid_controller', 'omni_drones.controllers.lee_position_controller', 'omni_drones.controllers', 'omni_drones.utils.torchrl.transforms'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Starting kit application with the following args:  ['/home/xingxun/isaac_sim/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/home/xingxun/experment/NavRL/isaac-training/third_party/orbit/_isaac_sim/apps/omni.isaac.sim.python.kit', '--/app/tokens/exe-path=/home/xingxun/experment/NavRL/isaac-training/third_party/orbit/_isaac_sim/kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True', '--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=True', '--/app/fastShutdown=True', '--ext-folder', '/home/xingxun/experment/NavRL/isaac-training/third_party/orbit/_isaac_sim/exts', '--ext-folder', '/home/xingxun/experment/NavRL/isaac-training/third_party/orbit/_isaac_sim/apps', '--/physics/cudaDevice=0', '--portable']
Passing the following args to the base kit application:  []
[Info] [carb] Logging to file: /home/xingxun/isaac_sim/kit/logs/Kit/Isaac-Sim/2023.1/kit_20250715_142135.log
2025-07-15 06:21:35 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[0.029s] [ext: omni.kit.async_engine-0.0.0] startup
[2025-07-15 14:21:35,410][asyncio][DEBUG] - Using selector: EpollSelector
[0.163s] [ext: omni.activity.core-1.0.1] startup
[0.165s] [ext: omni.assets.plugins-0.0.0] startup
[0.166s] [ext: omni.stats-0.0.0] startup
[0.166s] [ext: omni.client-1.0.1] startup
[0.170s] [ext: omni.activity.profiler-1.0.2] startup
[0.171s] [ext: omni.gpu_foundation-0.0.0] startup
[0.176s] [ext: omni.rtx.shadercache.vulkan-1.0.0] startup
[0.176s] [ext: carb.windowing.plugins-1.0.0] startup
[0.182s] [ext: omni.kit.renderer.init-0.0.0] startup

|---------------------------------------------------------------------------------------------|
| Driver Version: 535.183.01    | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 4060 Laptop.. | Yes: 0 |     | 8188    MB | 10de      | 0          |
|     |                                  |        |     |            | 28e0      | a72b0d4b.. |
|     |                                  |        |     |            | 1         |            |
|=============================================================================================|
| OS: 20.04.6 LTS (Focal Fossa) ubuntu, Version: 20.04.6, Kernel: 5.15.0-127-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12013000 (1.20.13.0)
| Processor: 13th Gen Intel(R) Core(TM) i9-13900HX | Cores: 24 | Logical: 24
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 31892 | Free Memory: 22825
| Total Page/Swap (MB): 2047 | Free Page/Swap: 2015
|---------------------------------------------------------------------------------------------|
[0.687s] [ext: omni.kit.pipapi-0.0.0] startup
[0.688s] [ext: omni.kit.pip_archive-0.0.0] startup
[0.689s] [ext: omni.pip.compute-1.2.0] startup
[0.689s] [ext: omni.pip.torch-2_0_1-2.0.2] startup
[0.701s] [ext: omni.pip.cloud-1.0.1] startup
[0.716s] [ext: omni.isaac.core_archive-2.2.1] startup
[0.717s] [ext: omni.isaac.ml_archive-1.1.3] startup
[0.717s] [ext: omni.kit.loop-isaac-1.1.0] startup
[0.717s] [ext: omni.kit.test-0.0.0] startup
[0.718s] [ext: omni.kit.telemetry-0.5.0] startup
[2025-07-15 14:21:35,978][omni.kit.telemetry.impl.sentry_extension][INFO] - sentry is disabled for external build
[2025-07-15 14:21:35,978][omni.kit.telemetry.impl.sentry_extension][INFO] - sentry is disabled for external build
[0.731s] [ext: omni.mtlx-0.1.0] startup
[0.732s] [ext: omni.usd.config-1.0.3] startup
[0.733s] [ext: omni.gpucompute.plugins-0.0.0] startup
[0.734s] [ext: omni.usd.libs-1.0.0] startup
[0.814s] [ext: omni.appwindow-1.1.5] startup
[1.468s] [ext: omni.kit.renderer.core-0.0.0] startup
[3.615s] [ext: omni.kit.renderer.capture-0.0.0] startup
[3.617s] [ext: omni.kit.renderer.imgui-0.0.0] startup
[3.670s] [ext: carb.audio-0.1.0] startup
[3.691s] [ext: omni.ui-2.18.1] startup
[3.707s] [ext: omni.uiaudio-1.0.0] startup
[3.708s] [ext: omni.kit.mainwindow-1.0.1] startup
[3.710s] [ext: omni.kit.uiapp-0.0.0] startup
[3.710s] [ext: omni.usd.schema.physics-0.0.0] startup
[... cut ...]
2025-07-15 06:29:28 [473,464ms] [Warning] [carb] Recursive unloadAllPlugins() detected!
2025-07-15 06:29:28 [473,470ms] [Warning] [omni.core.ITypeFactory] Module /home/xingxun/isaac_sim/kit/exts/omni.activity.core/bin/libomni.activity.core.plugin.so remained loaded after unload request.
/home/xingxun/miniconda3/envs/NavRL/lib/python3.10/multiprocessing/resource_tracker.py:224: UserWarning: resource_tracker: There appear to be 6 leaked semaphore objects to clean up at shutdown
  warnings.warn('resource_tracker: There appear to be %d '

```