Zhefan-Xu/NavRL #105: cd NavRL/isaac-training bash setup.sh这一步有问题

hao551, 2025-07-29:
调试时发现是这一步卡住了：echo "Running orbit.sh setup..."
./orbit.sh --conda $ENV_NAME
echo "hello"
单独运行orbit.sh出现
third_party/orbit/_isaac_sim/kit/plugins/carb_gfx:/home/robot/NavRL/isaac-training/third_party/orbit/_isaac_sim/kit/plugins/rtx:/home/robot/NavRL/isaac-training/third_party/orbit/_isaac_sim/kit/plugins/gpu.foundation' ''
+ echo -e '[INFO] Installing extra dependencies (this might take a few minutes)...'
[INFO] Installing extra dependencies (this might take a few minutes)...
+ conda install -c conda-forge -y importlib_metadata

hao551, 2025-07-29:
ubuntu20.04系统

hao551, 2025-07-29:
[image]
截图

skye-rookie, 2025-08-10:
我也遇到了同样的问题，请问您解决了吗

hao551, 2025-08-10:
没有呢，挺烦的哈哈哈哈

xunCai84, 2025-08-14:
请尝试修改`~/NavRL/isaac-training/third_party/orbit/orbit.sh`，将脚本160行中的`conda install -c conda-forge -y importlib_metadata &> /dev/null`修改为`PYTHONPATH="" conda install -c conda-forge -y importlib_metadata &> /dev/null`，执行`conda remove --name NavRL --all`删除虚拟环境，最后再尝试`bash setup.sh`应该可以解决。

对此的解释是您电脑中基础的 Conda 环境 (/home/users/anaconda3/) 与 Isaac Sim 自带的 Python 包 (.../_isaac_sim/exts/...) 发生了冲突。当 orbit.sh 脚本运行时，它为了能与 Isaac Sim 交互，会将 Isaac Sim 的内部库路径添加到了您的环境中。conda 命令本身是一个 Python 程序，它需要调用网络相关的库（如 OpenSSL, boto3, urllib3）来下载包。由于环境被“污染”，Conda 在尝试加载这些库时，错误地加载了 Isaac Sim 自带的、不兼容的旧版本库（从错误日志中的路径可以看出，它加载了 _isaac_sim/exts/.../boto3 等）。
最终，Isaac Sim 的库和您主 Conda 环境的库版本不匹配，导致了 AttributeError: module 'lib' has no attribute 'OpenSSL_add_all_algorithms' 的致命错误。

hao551, 2025-08-15:
谢谢，但是后面运行训练脚本时出现了X11问题，想降级到libxb1.15还不行

20190325lucky, 2025-12-27:
> 我也遇到了同样的问题，请问您解决请问你解决吗？我也是这个问题
、

experi12, 2026-02-03:
> 没有呢，挺烦的哈哈哈哈
> 
> skye-rookie ***@***.***>于2025年8月10日 周日15:02写道：
> […](#)

您好，请问这个问题解决了吗，困住一周了。

experi12, 2026-02-03:
> 我也遇到了同样的问题，请问您解决了吗
您好，请问这个问题解决了吗，困住一周了。