Zhefan-Xu/NavRL #116: 运行 train.py报错 怎么解决啊

Sa1toH, 2025-09-08:
/home/djq/.local/lib/python3.10/site-packages/torchrl/data/replay_buffers/samplers.py:36: UserWarning: Failed to import torchrl C++ binaries. Some modules (eg, prioritized replay buffers) may not work with your installation. This is likely due to a discrepancy between your package version and the PyTorch version. Make sure both are compatible. Usually, torchrl majors follow the pytorch majors within a few days around the release. For instance, TorchRL 0.5 requires PyTorch 2.4.0, and TorchRL 0.6 requires PyTorch 2.5.0.
  warnings.warn(EXTENSION_WARNING)
[Warning] [omni.isaac.kit.simulation_app] Modules: ['omni_drones', 'omni_drones.utils', 'omni_drones.utils.torchrl.env', 'omni_drones.utils.torchrl.collector', 'omni_drones.utils.torchrl', 'omni_drones.utils.torch', 'omni_drones.controllers.dsl_pid_controller', 'omni_drones.controllers.lee_position_controller', 'omni_drones.controllers', 'omni_drones.utils.torchrl.transforms'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Starting kit application with the following args:  ['/home/djq/isaac-sim-data/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py', '/home/djq/catkin_ws_NavRL/src/NavRL-main/isaac-training/third_party/orbit/_isaac_sim/apps/omni.isaac.sim.python.kit', '--/app/tokens/exe-path=/home/djq/catkin_ws_NavRL/src/NavRL-main/isaac-training/third_party/orbit/_isaac_sim/kit', '--/persistent/app/viewport/displayOptions=3094', '--/rtx/materialDb/syncLoads=True', '--/rtx/hydra/materialSyncLoads=True', '--/omni.kit.plugin/syncUsdLoads=True', '--/app/renderer/resolution/width=1280', '--/app/renderer/resolution/height=720', '--/app/window/width=1440', '--/app/window/height=900', '--/renderer/multiGpu/enabled=True', '--/app/fastShutdown=True', '--ext-folder', '/home/djq/catkin_ws_NavRL/src/NavRL-main/isaac-training/third_party/orbit/_isaac_sim/exts', '--ext-folder', '/home/djq/catkin_ws_NavRL/src/NavRL-main/isaac-training/third_party/orbit/_isaac_sim/apps', '--/physics/cudaDevice=0', '--portable']
Passing the following args to the base kit application:  []
[Info] [carb] Logging to file: /home/djq/isaac-sim-data/kit/logs/Kit/Isaac-Sim/2023.1/kit_20250908_112723.log
2025-09-08 03:27:23 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
[0.027s] [ext: omni.kit.async_engine-0.0.0] startup
[2025-09-08 11:27:23,537][asyncio][DEBUG] - Using selector: EpollSelector
[0.166s] [ext: omni.activity.core-1.0.1] startup
[0.168s] [ext: omni.assets.plugins-0.0.0] startup
[0.169s] [ext: omni.stats-0.0.0] startup
[0.169s] [ext: omni.client-1.0.1] startup
[0.173s] [ext: omni.activity.profiler-1.0.2] startup
[0.174s] [ext: omni.gpu_foundation-0.0.0] startup
[0.178s] [ext: omni.rtx.shadercache.vulkan-1.0.0] startup
[0.179s] [ext: carb.windowing.plugins-1.0.0] startup
[0.182s] [ext: omni.kit.renderer.init-0.0.0] startup
2025-09-08 03:27:23 [240ms] [Error] [carb.graphics-vulkan.plugin] VkResult: ERROR_INCOMPATIBLE_DRIVER
2025-09-08 03:27:23 [240ms] [Error] [carb.graphics-vulkan.plugin] vkCreateInstance failed. Vulkan 1.1 is not supported, or your driver requires an update.
2025-09-08 03:27:23 [240ms] [Error] [gpu.foundation.plugin] carb::graphics::createInstance failed.
2025-09-08 03:27:23 [405ms] [Error] [carb.graphics-vulkan.plugin] VkResult: ERROR_INCOMPATIBLE_DRIVER
2025-09-08 03:27:23 [405ms] [Error] [carb.graphics-vulkan.plugin] vkCreateInstance failed. Vulkan 1.1 is not supported, or your driver requires an update.
2025-09-08 03:27:23 [405ms] [Error] [gpu.foundation.plugin] carb::graphics::createInstance failed.
2025-09-08 03:27:23 [545ms] [Error] [omni.gpu_foundation_factory.plugin] Failed to create GPU foundation devices for compatibilityMode!
[0.552s] [ext: omni.kit.pipapi-0.0.0] startup
[0.553s] [ext: omni.kit.pip_archive-0.0.0] startup
[0.553s] [ext: omni.pip.compute-1.2.0] startup
[0.553s] [ext: omni.pip.torch-2_0_1-2.0.2] startup
[0.566s] [ext: omni.pip.cloud-1.0.1] startup
[0.568s] [ext: omni.isaac.core_archive-2.2.1] startup
[0.568s] [ext: omni.isaac.ml_archive-1.1.3] startup
[0.568s] [ext: omni.kit.loop-isaac-1.1.0] startup
[0.568s] [ext: omni.kit.test-0.0.0] startup
[0.569s] [ext: omni.kit.telemetry-0.5.0] startup
[xcb] Unknown sequence number while appending request
[xcb] You called XInitThreads, this is not your fault
[xcb] Aborting, sorry about that.
[xcb] Extra reply data still left in queue
[xcb] This is most likely caused by a broken X extension library
[xcb] Aborting, sorry about that.
Fatal Python error: Segmentation fault

Thread 0x00007fe072548180 (most recent call first):
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 1176 in create_module
  File "<frozen importlib._bootstrap>", line 571 in module_from_spec
  File "<frozen importlib._bootstrap>", line 674 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1006 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1027 in _find_and_load
  File "/home/djq/isaac-sim-data/kit/exts/omni.kit.telemetry/omni/kit/telemetry/__init__.py", line 7 in <module>
  File "<frozen importlib._bootstrap>", line 241 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 883 in exec_module
  File "<frozen importlib._bootstrap>", line 688 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1006 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1027 in _find_and_load
  File "<frozen importlib._bootstrap>", line 1050 in _gcd_import
  File "/home/djq/anaconda3/envs/NavRL/lib/python3.10/importlib/__init__.py", line 126 in import_module
  File "/home/djq/isaac-sim-data/kit/kernel/py/omni/ext/_impl/custom_importer.py", line 76 in import_module
  File "/home/djq/isaac-sim-data/kit/kernel/py/omni/ext/_impl/_internal.py", line 94 in _import_public
  File "/home/djq/isaac-sim-data/kit/kernel/py/omni/ext/_impl/_internal.py", line 187 in _custom_importer
  File "/home/djq/isaac-sim-data/kit/kernel/py/omni/ext/_impl/_internal.py", line 211 in startup
  File "/home/djq/isaac-sim-data/kit/kernel/py/omni/ext/_impl/_internal.py", line 328 in startup_extension
  File "PythonExtension.cpp::startup()", line 2 in <module>
  File "/home/djq/isaac-sim-data/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py", line 303 in _start_app
  File "/home/djq/isaac-sim-data/exts/omni.isaac.kit/omni/isaac/kit/simulation_app.py", line 192 in __init__
  File "/home/djq/catkin_ws_NavRL/src/NavRL-main/isaac-training/training/scripts/train.py", line 24 in main
  File "/home/djq/anaconda3/envs/NavRL/lib/python3.10/site-packages/hydra/core/utils.py", line 186 in run_job
  File "/home/djq/anaconda3/envs/NavRL/lib/python3.10/site-packages/hydra/_internal/hydra.py", line 119 in run
  File "/home/djq/anaconda3/envs/NavRL/lib/python3.10/site-packages/hydra/_internal/utils.py", line 458 in <lambda>
  File "/home/djq/anaconda3/envs/NavRL/lib/python3.10/site-packages/hydra/_internal/utils.py", line 220 in run_and_report
  File "/home/djq/anaconda3/envs/NavRL/lib/python3.10/site-packages/hydra/_internal/utils.py", line 457 in _run_app
  File "/home/djq/anaconda3/envs/NavRL/lib/python3.10/site-packages/hydra/_internal/utils.py", line 394 in _run_hydra
  File "/home/djq/anaconda3/envs/NavRL/lib/python3.10/site-packages/hydra/main.py", line 94 in decorated_main
  File "/home/djq/catkin_ws_NavRL/src/NavRL-main/isaac-training/training/scripts/train.py", line 138 in <module>

Extension modules: yaml._yaml, numpy.core._multiarray_umath, numpy.core._multiarray_tests, numpy.linalg._umath_linalg, numpy.fft._pocketfft_internal, numpy.random._common, numpy.random.bit_generator, numpy.random._bounded_integers, numpy.random._mt19937, numpy.random.mtrand, numpy.random._philox, numpy.random._pcg64, numpy.random._sfc64, numpy.random._generator, h5py._errors, h5py.defs, h5py._objects, h5py.h5, h5py.utils, h5py.h5t, h5py.h5s, h5py.h5ac, h5py.h5p, h5py.h5r, h5py._proxy, h5py._conv, h5py.h5z, h5py.h5a, h5py.h5d, h5py.h5ds, h5py.h5g, h5py.h5i, h5py.h5o, h5py.h5f, h5py.h5fd, h5py.h5pl, h5py.h5l, h5py._selector, google.protobuf.pyext._message, psutil._psutil_linux, psutil._psutil_posix, torch._C, torch._C._fft, torch._C._linalg, torch._C._nested, torch._C._nn, torch._C._sparse, torch._C._special (total: 48)
段错误 (核心已转储)

ablearner, 2025-11-21:
同问

imagic79, 2026-03-05:
解决方法   在你的终端中确保当前处于 NavRL 虚拟环境下，然后尝试移除 Conda 环境中易冲突的 libxcb 和 xorg-libx11 组件（让系统回退使用 Ubuntu 原生的库）：   conda remove libxcb xorg-libx11